Ai-Thinker-Open/GPRS_C_SDK #1: coolwatcher 下载程序失败

chenxiuqimr, 2017-11-25:
新买的A9G开发板，下载时提示如下，无法下载呢？

Event sniffer (re)starting. (Connection: Event Sniffer (COM3))
Loading profile script : 8955.rb
EVENT SNIFFING ERROR: CONNECTION BROKEN?
EVENT SNIFFING RESURRECTED.
> fastpfChooseFlashProgrammer()
Flash programmer file for fastpfization set to: 
D:/Project/GPRS_A9/GPRS_C_SDK/platform/ramrun/flsh_spi32m/flsh_spi32m_CUSTOMER_host_ramrun.lod
> fastpfChooseLod()
Lod file for fastpfization set to: 
D:/Project/GPRS_A9/GPRS_C_SDK/hex/gpio_B1003.lod
> fastpfGo()
Fastpf V2 over Host
Loading the lod files: 
gpio_B1003.lod
Using flash programmer: 
flsh_spi32m_CUSTOMER_host_ramrun.lod
Ramrunning the flash programmer...

Resetting SPI flash...
Done.

Entered Host Monitor mode.

No boot sector struct exists. Skipping ...

0.141000 0.000000 0.141000 ( 0.901335)

Version mismatch between XMD and embedded flash programmer! (embedded version: fa01.fb04 vs script version: fa01.fb05)
It should be compatible, but it is recommended to match XMD and embedded flash programmer
Verify enabled: true
Fastpf Protocol Version: 1.5
Got FPC buffer size: 32768
Use FPC buffer size: 32768
Boot sector address: 0x0
Fastpfing...
Event sniffer (re)starting. (Connection: Event Sniffer (COM3))

￼ `waitCommandResult': Received unknown command event 0xd0 from target for command 0xf0 (FastPf::FlashCommandTriggeredError)

./chipgen/Modem2G/toolpool/plugins/fastpf/fastpf.rb   67   `waitCommandResult'
./chipgen/Modem2G/toolpool/plugins/fastpf/fastpf.rb   309   `dofastpfXfer'
./chipgen/Modem2G/toolpool/plugins/fastpf/fastpf.rb   306   `each'
./chipgen/Modem2G/toolpool/plugins/fastpf/fastpf.rb   306   `dofastpfXfer'
D:/Project/GPRS_A9/CSDTK/cooltools/lib/ruby/1.8/benchmark.rb   293   `measure'
./chipgen/Modem2G/toolpool/plugins/fastpf/fastpf.rb   305   `dofastpfXfer'
./chipgen/Modem2G/toolpool/plugins/fastpf/fastpf.rb   718   `dofastpf'
./chipgen/Modem2G/toolpool/plugins/fastpf/fastpf.rb   788   `fastpf'
./chipgen/Modem2G/toolpool/plugins/fastpf/fastpfplugin.rb   83   `fastpfGenericGo'
./chipgen/Modem2G/toolpool/plugins/fastpf/fastpfplugin.rb   112   `fastpfGo'
EVENT SNIFFING ERROR: CONNECTION BROKEN?
EVENT SNIFFING RESURRECTED.

Neutree, 2017-11-25:
参照[这里](https://github.com/Ai-Thinker-Open/GPRS_C_SDK/blob/master/doc/download_debug_tool_zh-cn.md#coolwatcher无法下载)，看有没有相同的情况

chenxiuqimr, 2017-11-25:
仔细对照你给的参考了，试了按照教程下载at指令固件都能正常下载。二次开发安装教程编译完gpio例程都能正常编译，只是到了coolwatcher 下载这里不行，试了很多次，而且也试了重启板子待通信指示灯闪时下载同样，一直提示。￼ `waitCommandResult': Received unknown command event 0xd0 from target for command 0xf0 (FastPf::FlashCommandTriggeredError)

Neutree, 2017-11-25:
哦哦，问题出在ramrun文件选择错误了，原因是`GPRSC_SDK`中的`platform/ramrun`目录下的文件版本比coolwatcher软件版本低，所以解决方法：
[image]

使用`CSDTK/cooltools\chipgen\Modem2G\toolpool\plugins\fastpf\flash_programmers`下的名字包含8955 ramrun spi32m .lod的文件就可以了，抱歉了，更新了coolwatcher没有更新SDK里面的文件！

closed 2017-11-25 by Neutree

zzs123, 2018-03-15:
Connection $CURRENTCONNECTION (COM1) opened (#CURRENTCONNECTION).
--

Tianxiaomo, 2018-08-24:
Hi，I use CSDTK42,but don't download.
This is error message

￼ `CHST_Read32': Request to coolhost timed out ! (CoolHostTimeoutError)

./rbbase/common/coolhost_connections.rb
161
`CHST_Read32'
./rbbase/common/coolhost_connections.rb
161
`read32'
./rbbase/common/base.rb
85
`read'
./rbbase/common/base.rb
239
`read'
./chipgen/Modem2G/toolpool/scripts/8809/target_executor.rb
72
`enterHostMonitor'
./chipgen/Modem2G/toolpool/scripts/ChipStd/target_executor.rb
366
`session'
./chipgen/Modem2G/toolpool/scripts/ChipStd/ramrun.rb
12
`doramrunXfer'
./chipgen/Modem2G/toolpool/scripts/ChipStd/ramrun.rb
62
`ramrunXfer'
./chipgen/Modem2G/toolpool/plugins/fastpf/fastpf.rb
212
`dofastpfXfer'
./chipgen/Modem2G/toolpool/plugins/fastpf/fastpf.rb
718
`dofastpf'
./chipgen/Modem2G/toolpool/plugins/fastpf/fastpf.rb
788
`fastpf'
./chipgen/Modem2G/toolpool/plugins/fastpf/fastpfplugin.rb
83
`fastpfGenericGo'
./chipgen/Modem2G/toolpool/plugins/fastpf/fastpfplugin.rb
112
`fastpfGo'

EVENT SNIFFING ERROR: CONNECTION BROKEN?
EVENT SNIFFING RESURRECTED.

Johhnzero, 2019-04-12:
I have the same problem

psatya111, 2019-04-27:
I am facing the same problem .please check the below logs:
Detected event: 0x66660021.
Detected event: 0x0000000f.
Detected event: 0x00000000.
Detected event: 0x001660c8.
Detected system reset (0x57).
Detected event: 0xca1bc2c0.
Detected event: 0x9999a244.
Detected event: 0x77770006.
Detected event: 0xffffff00.
Detected event: 0xb0070000.
> fastpfGo()
Fastpf V2 over Host
Loading the lod files: 
oled_B2129_debug.lod
Using flash programmer: 
host_8955_flsh_spi32m_ramrun.lod
Ramrunning the flash programmer...
Event sniffer (re)starting. (Connection: Event Sniffer (COM1))
undefined method `CHST_Read32' for #<CHHostConnection:0x7fb9c36c4bc8> (NoMethodError)

./rbbase/common/base.rb | 85 | `read'
-- | -- | --
./rbbase/common/base.rb | 239 | `read'
./chipgen/Modem2G/toolpool/scripts/8809/target_executor.rb | 72 | `enterHostMonitor'
./chipgen/Modem2G/toolpool/scripts/ChipStd/target_executor.rb | 366 | `session'
./chipgen/Modem2G/toolpool/scripts/ChipStd/ramrun.rb | 12 | `doramrunXfer'
./chipgen/Modem2G/toolpool/scripts/ChipStd/ramrun.rb | 62 | `ramrunXfer'
./chipgen/Modem2G/toolpool/plugins/fastpf/fastpf.rb | 212 | `dofastpfXfer'
./chipgen/Modem2G/toolpool/plugins/fastpf/fastpf.rb | 718 | `dofastpf'
./chipgen/Modem2G/toolpool/plugins/fastpf/fastpf.rb | 788 | `fastpf'
./chipgen/Modem2G/toolpool/plugins/fastpf/fastpfplugin.rb | 83 | `fastpfGenericGo'
./chipgen/Modem2G/toolpool/plugins/fastpf/fastpfplugin.rb | 112 | `fastpfGo'

Need help to resolve this.please please please help...

xuehu96, 2020-04-11:
试了CSDTK42和GPRS_C_SDK下面的ramrun，都不能成功下载。

===============
Flash programmer file for fastpfization set to: 
E:/A9Dev/CSDTK42/cooltools/chipgen/Modem2G/toolpool/plugins/fastpf/flash_programmers/host_8955_flsh_spi32m_ramrun.lod
> fastpfGo()
Fastpf V2 over Host
Loading the lod files: 
gpio_B2129_debug.lod
Using flash programmer: 
host_8955_flsh_spi32m_ramrun.lod
Ramrunning the flash programmer...
Event sniffer (re)starting. (Connection: Event Sniffer (COM3))

￼ `CHST_Read32': Request to coolhost timed out ! (CoolHostTimeoutError)

============
> fastpfChooseFlashProgrammer()
Flash programmer file for fastpfization set to: 
E:/A9Dev/GPRS_C_SDK/platform/ramrun/flsh_spi32m/host_8955_flsh_spi32m_ramrun.lod
> fastpfGo()
Fastpf V2 over Host
Loading the lod files: 
gpio_B2129_debug.lod
Using flash programmer: 
host_8955_flsh_spi32m_ramrun.lod
Ramrunning the flash programmer...
Event sniffer (re)starting. (Connection: Event Sniffer (COM3))

￼ `CHST_Read32': Request to coolhost timed out ! (CoolHostTimeoutError)

natrayanp, 2020-09-07:
Getting the same error in Linux:
> fastpfGo()
Fastpf V2 over Host
Loading the lod files: 
sms_B2129_debug.lod
Using flash programmer: 
host_8955_flsh_spi32m_ramrun.lod
Ramrunning the flash programmer...
Event sniffer (re)starting. (Connection: Event Sniffer (COM1))
undefined method `CHST_Read32' for #<CHHostConnection:0x7f768d4792c0> (NoMethodError)
--

./rbbase/common/base.rb | 85 | `read'
-- | -- | --
./rbbase/common/base.rb | 239 | `read'
./chipgen/Modem2G/toolpool/scripts/8809/target_executor.rb | 72 | `enterHostMonitor'
./chipgen/Modem2G/toolpool/scripts/ChipStd/target_executor.rb | 366 | `session'
./chipgen/Modem2G/toolpool/scripts/ChipStd/ramrun.rb | 12 | `doramrunXfer'
./chipgen/Modem2G/toolpool/scripts/ChipStd/ramrun.rb | 62 | `ramrunXfer'
./chipgen/Modem2G/toolpool/plugins/fastpf/fastpf.rb | 212 | `dofastpfXfer'
./chipgen/Modem2G/toolpool/plugins/fastpf/fastpf.rb | 718 | `dofastpf'
./chipgen/Modem2G/toolpool/plugins/fastpf/fastpf.rb | 788 | `fastpf'
./chipgen/Modem2G/toolpool/plugins/fastpf/fastpfplugin.rb | 83 | `fastpfGenericGo'
./chipgen/Modem2G/toolpool/plugins/fastpf/fastpfplugin.rb | 112 | `fastpfGo'

EVENT SNIFFING ERROR: CONNECTION BROKEN?
EVENT SNIFFING RESURRECTED.

Richelfo, 2023-02-23:
Hi Guys.
Any of you could finally solve this.  I have the same problem. Using Windows 11, prolific cable PL2303 with updated driver, Can´t download any file to A9G

Tad3a, 2023-07-20:
Same problem

MuhammetCakirr, 2023-10-01:
same problem is there any solution

tadeastehan, 2024-12-29:
https://github.com/Ai-Thinker-Open/GPRS_C_SDK/issues/140