Ai-Thinker-Open/GPRS_C_SDK #6: coolwatch下载出现问题

sqm289798469, 2017-12-05:
参照文档里的问题，试过豆不行，ramrun文件换过也不行

sqm289798469, 2017-12-05:
> fastpfGo()
Fastpf V2 over Host
Loading the lod files: 
gps_B1003.lod
Using flash programmer: 
host_8955_flsh_spi32m_ramrun.lod
Ramrunning the flash programmer...
Event sniffer (re)starting. (Connection: Event Sniffer (COM4))

￼ `CHST_Read32': Request to coolhost timed out ! (CoolHostTimeoutError)

./rbbase/common/coolhost_connections.rb
161
`CHST_Read32'
./rbbase/common/coolhost_connections.rb
161
`read32'
./rbbase/common/base.rb
85
`read'
./rbbase/common/base.rb
239
`read'
./chipgen/Modem2G/toolpool/scripts/8809/target_executor.rb
72
`enterHostMonitor'
./chipgen/Modem2G/toolpool/scripts/ChipStd/target_executor.rb
366
`session'
./chipgen/Modem2G/toolpool/scripts/ChipStd/ramrun.rb
12
`doramrunXfer'
./chipgen/Modem2G/toolpool/scripts/ChipStd/ramrun.rb
62
`ramrunXfer'
./chipgen/Modem2G/toolpool/plugins/fastpf/fastpf.rb
212
`dofastpfXfer'
./chipgen/Modem2G/toolpool/plugins/fastpf/fastpf.rb
718
`dofastpf'
./chipgen/Modem2G/toolpool/plugins/fastpf/fastpf.rb
788
`fastpf'
./chipgen/Modem2G/toolpool/plugins/fastpf/fastpfplugin.rb
83
`fastpfGenericGo'
./chipgen/Modem2G/toolpool/plugins/fastpf/fastpfplugin.rb
112
`fastpfGo'

EVENT SNIFFING ERROR: CONNECTION BROKEN?
EVENT SNIFFING RESURRECTED.

Neutree, 2017-12-05:
这种情况，请优先检查硬件连接，保证USB转串口模块的`TX`和`RX`引脚连接的是开发板（/模块）的`HST_RX`和`HST_TX`引脚，主要交叉连接，然后检查软件设置
[image]

Neutree, 2017-12-05:
需要注意的是天线不要碰到引脚
[image]

closed 2017-12-05 by Neutree

Yachara, 2019-05-21:
I had the same problem.
Was using PL-2303HX. Windows  was reporting a problem:

Warning Notice:
Please be warned that counterfeit (fake) PL-2303HX (Chip Rev A) USB to Serial Controller ICs using Prolific's trademark logo, brandname, and device drivers, were being sold in the China market. Counterfeit IC products show exactly the same outside chip markings but generally are of poor quality and causes Windows driver compatibility issues (Yellow Mark Error Code 10 in Device Manager).

Switched to CP2102, everything worked first try.