Ai-Thinker-Open/GPRS_C_SDK #7: demo.gps源码程序问题

sqm289798469, 2017-12-05:
烧录gps的源码后，串口连续接收到信息，如果想每隔一段时间发送，是否只需再while里加入os_sleep
    while(1)
    {
        if(OS_WaitEvent(gpsTaskHandle, &event, OS_TIME_OUT_WAIT_FOREVER))
        {
            EventDispatch(event);
            OS_Free(event->pParam1);
            OS_Free(event);
        }
    }

Neutree, 2017-12-05:
如果是**主任务**，主任务里面的这个循环内的OS_WaitEvent()是必须的，而且这个函数是阻塞的，主任务主要用来处理事件。
**不建议在这个任务里面写除了处理事件的代码。**
如果你要用OS_Sleep()来循环等待发送的话你可再建立一个任务，在那个任务里这样做是可以的，或者用定时器也可以

sqm289798469, 2017-12-06:
定时器写法和单片机一样吗

Neutree, 2017-12-06:
定时器封装到OS中了，即调用一个函数就行了，很方便简单。

sqm289798469, 2017-12-06:
不太明白，os_sleep()相当于单片机的delay（）延时的话，定时函数时哪个？可以举个例子吗

Neutree, 2017-12-06:
定时函数应该是 `OS_StartCallbackTimer(...)`
刚刚看了下，目前版本的还没加进来，下次更新的时候一起更新进来
你可以先用新建一个任务用死循环+延时的方式，新建任务的方式和主函数中建立主任务的方法一样，调用`OS_CreateTask(...)`建立新的任务即可

sqm289798469, 2017-12-06:
我在demo里添加了新任务及循环，串口还是没有延时输出

#include <string.h>
#include <stdio.h>
#include <api_os.h>
#include <api_gps.h>
#include <api_event.h>
#include <api_hal_uart.h>
#include <api_debug.h>

#define MAIN_TASK_STACK_SIZE    (2048 * 2)
#define MAIN_TASK_PRIORITY      0
#define MAIN_TASK_NAME          "GPS Test Task"

static HANDLE gpsTaskHandle = NULL;
static HANDLE sendwait = NULL;

void EventDispatch(API_Event_t* pEvent)
{
    switch(pEvent->id)
    {
        case API_EVENT_ID_GPS_UART_RECEIVED:
//             Trace(1,"received GPS data,length:%d, data:%s",pEvent->param1,pEvent->pParam1);
            UART_Write(UART1,pEvent->pParam1,pEvent->param1);
            break;
        case API_EVENT_ID_UART_RECEIVED:
            if(pEvent->param1 == UART1)
            {
                uint8_t data[pEvent->param2+1];
                data[pEvent->param2] = 0;
                memcpy(data,pEvent->pParam1,pEvent->param2);
                Trace(1,"uart received data,length:%d,data:%s",pEvent->param2,data);
            }
            break;
        default:
            break;
    }
}

void gps_MainTask(void *pData)
{
    API_Event_t* event=NULL;

    GPS_Open(NULL);
    UART_Config_t config = {
        .baudRate = UART_BAUD_RATE_115200,
        .dataBits = UART_DATA_BITS_8,
        .stopBits = UART_STOP_BITS_1,
        .parity   = UART_PARITY_NONE,
        .rxCallback = NULL,
        .useEvent   = true
    };
    UART_Init(UART1,config);

    while(1)
    {
        if(OS_WaitEvent(gpsTaskHandle, &event, OS_TIME_OUT_WAIT_FOREVER))
        {
            EventDispatch(event);
            OS_Free(event->pParam1);
            OS_Free(event);
        }
    }
}

void gps_send(void *pData)
{
    int i;
    while(1)
    {
        i++;
        if(i == 300)
        {
            i = 0;
            OS_Sleep(20000);
        }
    }
}
void gps_Main(void)
{
    gpsTaskHandle = OS_CreateTask(gps_MainTask,
        NULL, NULL, MAIN_TASK_STACK_SIZE, MAIN_TASK_PRIORITY, 0, 0, MAIN_TASK_NAME);
    sendwait = OS_CreateTask(gps_send,
        NULL, NULL, MAIN_TASK_STACK_SIZE, MAIN_TASK_PRIORITY, 0, 0, MAIN_TASK_NAME);
    OS_SetUserMainHandle(&gpsTaskHandle);
}

程序哪里需要改进

Neutree, 2017-12-06:
。。。。。我。。。

Neutree, 2017-12-06:
你循环里啥事儿都没做啊。。

Neutree, 2017-12-06:
请自己仔细阅读一遍自己的代码，仔细检查你的代码的逻辑。。。

closed 2017-12-06 by Neutree

sqm289798469, 2017-12-06:
事件处理EventDispatch(API_Event_t* pEvent)中不是已经获取GPS信息输出了吗？while里还要怎么写？

Neutree, 2017-12-06:
你想实现什么功能

sqm289798469, 2017-12-06:
简单啊，demo。gps是连续发送信息到串口，我只想每隔10秒往串口发送一次

Neutree, 2017-12-06:
那你就需要在 API_EVENT_ID_GPS_UART_RECEIVED 事件下面处理gps输出信息，
然后在循环中10s发送一次你想发的信息