Ai-Thinker-Open/GPRS_C_SDK #9: 下载时提示 No boot sector struct exists

notobject, 2017-12-06:
Fastpf V2 over Host
Loading the lod files: 
host_8955_flsh_spi32m_ramrun.lod
Using flash programmer: 
gpio_flash.lod
Ramrunning the flash programmer...

Resetting SPI flash...
Done.

Entered Host Monitor mode.

No boot sector struct exists. Skipping ...

Event sniffer (re)starting. (Connection: Event Sniffer (COM5))

Neutree, 2017-12-06:
这个不用理会，这是正常的，能下载运行就行

notobject, 2017-12-06:
右下角一直没有进度条,等了一会之后.错误提示如下:
 `dofastpfXfer': The flash programmer has not sent his READY event! (FastPf::FlashProgrammerNotReady)

./chipgen/Modem2G/toolpool/plugins/fastpf/fastpf.rb | 216 | `dofastpfXfer'
-- | -- | --
D:/software/CSDTK/cooltools/lib/ruby/1.8/benchmark.rb | 293 | `measure'
./chipgen/Modem2G/toolpool/plugins/fastpf/fastpf.rb | 214 | `dofastpfXfer'
./chipgen/Modem2G/toolpool/plugins/fastpf/fastpf.rb | 718 | `dofastpf'
./chipgen/Modem2G/toolpool/plugins/fastpf/fastpf.rb | 788 | `fastpf'
./chipgen/Modem2G/toolpool/plugins/fastpf/fastpfplugin.rb | 83 | `fastpfGenericGo'
./chipgen/Modem2G/toolpool/plugins/fastpf/fastpfplugin.rb | 112 | `fastpfGo'

notobject, 2017-12-06:
[image]
上面是完整的输出截图

notobject, 2017-12-06:
另外,我之前下载`gpio_blood.lod`成功过一次,所以排除了硬件连接和软件设置的问题.第二次下载`gpio_flash.lod`就出错了, 然后都下不了

Neutree, 2017-12-06:
进度条是在command窗口显示的
[image]

顺便看一下软件版本
[image]

closed 2017-12-27 by Neutree

brainbush, 2017-12-27:
你好，我这也遇到了一样的问题，在下载的时候command的右下角部分不会显示进度条，但是USB转串口设备的RX TX灯常亮，随后也提示这个错误，设备无任何响应，重置后任然是之前的固件，此时无论重新插拔USB转串口设备还是在CoolHost内重新连接串口（即使提示成功），CoolWatcher下载固件提示超时，只能重新打开CoolWatcher

已尝试操作：

- 更换USB转串口设备
- 在其他电脑上重新配置环境+编译+下载（均为最新版本）
- 使用另一块全新的A9G开发板
- checkout到昨天的v1.3 尝试编译+下载

不知道这些信息是否有帮助……

Neutree, 2017-12-27:
上电后USB转串口不应该常亮的，有数据通讯会闪烁，如果是常亮应该是RXTX接错了

Neutree, 2017-12-27:
一定要接HST_RX引脚和HST_TX引脚，如果确认线没有接错，
点击下载后看有没有如下提示
[image]

如果有这个提示表示下载已经开始了，不过在过程中断开了，可能是由于程序进入了死循环导致，可以在按板子上的复位按键后串口灯开始闪的时候点击下载程序就可以了

brainbush, 2017-12-27:
表示确认多次RX TX没有接反（因为连着的时候重新下载原版的AT指令固件的时候是没有问题的），下载的时候没有进入 ramrunning the flash programmer 这一步（用AT固件的时候使用coolwatcher更新也是如此）……

[image]

（在中间尝试的时候有一次成功了，但是再次重新下载gpio demo的时候出现一样的错误

Neutree, 2017-12-27:
噢噢，你这个已经进入了下载了，只是下到一般失败了，是不是在程序里面写了while(1){}，可以在按板子上的复位按键后串口灯开始闪的时候快速点击下载程序按钮应该就可以了

brainbush, 2017-12-27:
嗯，我也知道可能是死循环导致无法下载的问题。但是我这边在AT固件的基础上用没有修改的GPIO示例也会出现这问题，而且上电复位（包括重新连接串口）也不行……

重开coolwatcher后不用按下复位之类的也可以直接进入前面那张截图的状态，但是还是下载失败……一直没有头绪

Neutree, 2017-12-27:
咦，你用AT固件的升级工具可以下载么
http://wiki.ai-thinker.com/gprs_download

Neutree, 2017-12-27:
第一次下载需要下载hex目录下那个比较大的文件，大概超过4M的文件

brainbush, 2017-12-27:
嗯，能用AT下载工具下载afw后缀的固件，在coolwatcher内lod用的是4M的文件，dry用的是platform/csdk的文件，也是一样的情况……

Neutree, 2017-12-27:
你下载我编译的这个试试
[hex.zip](https://github.com/Ai-Thinker-Open/GPRS_C_SDK/files/1589392/hex.zip)

brainbush, 2017-12-27:
用了还是和之前一样……

brainbush, 2017-12-27:
等白天我再换个型号的USB转串口再试试吧……（试过两个CH340都是这样的情况）
幸苦了……

Neutree, 2017-12-27:
换个电源供电试试，可能下载途中断电了

brainbush, 2017-12-28:
你好，我这边换了CP1202以及配合外部USB供电试了一下，还是一样的错误……（还是AT固件能正常下载但coolwatcher下载失败）

Neutree, 2017-12-28:
# [gpio_B1400.zip](https://github.com/Ai-Thinker-Open/GPRS_C_SDK/files/1590576/gpio_B1400.zip)

看报错信息是由于软件版本和固件BootLoader版本不兼容导致。。。，先用升级工具下载这个然后再用ccoolwatcher下载试试

brainbush, 2017-12-28:
呃……好像无法下载的样子……
[image]
[image]

Neutree, 2017-12-28:
。。。。文件校验值没错
我先下载AT固件，然后再用coolwatcher也是可以下载的呀。。

**[Ai-Thinker_A9_firmware_release_v1.6_RC.zip](https://github.com/Ai-Thinker-Open/GPRS_C_SDK/files/1590809/Ai-Thinker_A9_firmware_release_v1.6_RC.zip)**

[image]

[image]

brainbush, 2017-12-28:
我这边按照所说的用1.6的固件重新下，终于成功了……
非常感谢！

Neutree, 2017-12-28:
真棒,总算是可以了...