Ai-Thinker-Open/GPRS_C_SDK #10: 关于A9G的SDK的API问题

Spring641, 2017-12-07:
请问目前SDK是否支持SPI接口、I2C2接口、ADC接口的函数，最主要的就是ADC功能的API，其他的可以用GPIO模拟都无所谓。在你们淘宝店看到的是写着支持这些功能。不过我在SDK里面没有找到相应的ADC的API，还有你们的论坛有关资料我都看过了。目前项目开发准备用这个模块当主控，还麻烦贵公司给个满意答复，谢谢！

Neutree, 2017-12-07:
已经发布的SDK中暂时没有，后面会加入的，ADC功能已经有了，这周有空了更新上来就可以用了

Spring641, 2017-12-07:
好的麻烦了，不过移植最新的sdk，那我之前添加的代码兼么，目前我这已经写了一部分应用

Neutree, 2017-12-08:
兼容的，对前面已经写好的代码没什么改动，有改动会说明的

Spring641, 2017-12-08:
在线等你们的下个版本。。。。

Neutree, 2017-12-08:
。。。。还是先睡觉觉吧。。。这两天有其它更紧急的事情。。所以sorry啦

Spring641, 2017-12-09:
[image]
[image]
[image]

Spring641, 2017-12-09:
这次我测试用SDK来解析GPS的数据，在上图1是解析函数移植到ESP8266里面运行，图2是本次移植到A9G开发板后的串口打印结果，图3是代码是在windows 7用啊哈C软件写的测试解析函数的运行结果。最后发现在windows系统运行和ESP8266模块运行解析GPS数据是没问题的，问题是在A9G模块运行自己写的解析函数时出现了问题，貌似A9G模块直接死掉了。问：是否这次的SDK是否存在用户代码量？或者有什么bug呢？个人感觉在sdk里面写代码很多结果都不是自己想要的。希望贵公司看看本人调试出现的问题。

Neutree, 2017-12-09:
从你的截图比较难定位问题。。。。

Spring641, 2017-12-12:
还在等你们的下个版本呢。。。。不是你们上传就可以了么，，，，，，。。。。。

closed 2017-12-27 by Neutree