Ai-Thinker-Open/GPRS_C_SDK #15: gps信息必须通过串口读出解析后再通过Socket_TcpipWrite发送到服务器上面吗

sqm289798469, 2017-12-12:
gps信息必须通过串口读出解析后再通过Socket_TcpipWrite发送到服务器上面吗

Neutree, 2017-12-12:
...这个都是功能上的逻辑了，，，看你自己有什么需求咯

sqm289798469, 2017-12-12:
Socket_TcpipWrite（套接字，数据，数据长度），
gps的demo里，触发gps时间，串口输出是 UART_Write(UART1,pEvent->pParam1,pEvent->param1);
,pEvent->pParam1,pEvent->param1这2个是否就是数据和长度，
对代码这一块真心不懂

Neutree, 2017-12-12:
。。。。对呀。。在api_event.h中也有说明
[image]

Neutree, 2017-12-12:
等会儿我写一个定时发送gps定位信息的demo吧。。。看你搞了这么久都没搞定！！！基础不牢哦，加油

Neutree, 2017-12-12:
静下心来多看看多谢谢多调试调试代码，大家都是这么一步一步过来的

sqm289798469, 2017-12-12:
嗯，谢谢，入门硬件这一块才一个多月，先前主要学arduino

sqm289798469, 2017-12-12:
[image]
这是信息丢失了吗？

Neutree, 2017-12-12:
https://github.com/Ai-Thinker-Open/GPRS_C_SDK/blob/master/demo/gps/src/demo_gps.c
写了个定时发送GPS原始信息到串口，可以作为参考

closed 2017-12-26 by Neutree