Ai-Thinker-Open/GPRS_C_SDK #24: ADC and POWER management support?

khaledbenh, 2017-12-21:
hello i would like to use your product in low power application using gps and gprs, i need to use power management modes to save power, and adc to read voltages of battery and of usb charging port, i didnt find out how to implement them from the sdk, is there support for the 2 ADC ports ? is there power modes and a way to switch between them ?

Neutree, 2017-12-21:
support in next release

khaledbenh, 2017-12-21:
can i know when is the next release ? will it be in some days ? a week ? a month ? just approximately 
can i contribute in the development of the next release ?

Neutree, 2017-12-21:
today or tomorror

khaledbenh, 2017-12-21:
thank you for your replay , you have a very good product, and your coding style is very comprehensive, waiting eagerly for the next release

Neutree, 2017-12-27:
[power manager](https://github.com/Ai-Thinker-Open/GPRS_C_SDK/blob/master/demo/pm/src/demo_pm.c)
[ADC](https://github.com/Ai-Thinker-Open/GPRS_C_SDK/blob/master/demo/adc/src/demo_adc.c)

closed 2017-12-27 by Neutree