Ai-Thinker-Open/GPRS_C_SDK #30: 请问什么时候出SSL的库？

lqq96, 2017-12-27:
请问什么时候出SSL的接口，期待对接MQTT和之后出FOTA。这样基本的商业化需求就满足了。

Neutree, 2017-12-27:
基本的MQTT已经可以了，SSL最近会调，
FOTA的话可以加入，但会稍微放到后面更新

lqq96, 2017-12-27:
有大致功能更新的时间表吗？

Neutree, 2017-12-27:
SSL争取在元旦后弄好，FOTA暂时还没排时间弄

ju-sti, 2017-12-31:
Hello Neutree, MQTT+SSL doesn't work for me. As soon as I enable SSL it doesn't connect anymore. It looks like there is a problem with DNS "NOT FOUND"? After the screenshot nothing with relation to mqtt happens anymore. I've checked the used settings with another mqtt client, they are valid. Thanks 
[image]

lqq96, 2018-01-21:
昨天我测试了你们集成的mqtt部分，对接的服务器为阿里物联网套件，mqtt+mbedtls，在测试你们的demo时候没有任何问题，可是在移植到阿里物联网套件的时候，就对接不上了。测试过看多次都不可以。首先你们的api不清晰，你们既然没有对外开放mqtt和tls源码，那么你们应该列出一个文档，告知每个api的用法和每个参数的定义。
其次，在调用SSL的时候，所有的证书不可以传空，传空会报错程序停止。希望可以有一个参数可以选择，选择只样子服务器证书或者验证服务器和客户端或者都不验证。使用过ci.ssl_verify_mode = 0;
并不好用。在普通的mqtt版本连接上了阿里套件，也可以正常收发消息，这很欣慰，在客户端这边发布消息的时候，会出现一个发送错误，-3，看是超时的错误代码，实时这个消息服务器是收到了的。
希望解决的问题：1 开放mqtt和ssl源码或者发布一份详细的api使用手册。demo并不能介绍每一个api的多种用例。
2 改善SSL的方法，不知道你们内部怎么调用的mbedtls，只是他们在内部验证的证书，希望可以只有选择验证那一部分证书，不选择验证的可以传空指针。
3 希望可以选择SSL版本，如果不支持的话，希望可以尽量高版本。
以上是我测试的结果和建议，如果有不合理的地方也请指出，谢谢。

Neutree, 2018-01-21:
嗯嗯，目前发布的版本不建议使用SSL，是不完善的，等下次更新SSL后再使用，你说的问题都已经解决了

Neutree, 2018-01-24:
ssl已经上传，可以尝试使用

ju-sti, 2018-01-28:
Hello neutree, I still have similar error. It doesn't show proper error message but crashes
[image]

Neutree, 2018-01-29:
@ju-sti 使用的是哪个版本的SDK呢，使用GDB调试提示什么呢

更新到V1.505版本应该能解决的问题

ju-sti, 2018-01-30:
With v1.505 the crash disappeared. Thank you.