Ai-Thinker-Open/GPRS_C_SDK #32: mqtt task error 中断执行Detected event: 0x9db00000

lappaport, 2017-12-27:
[22:06:55.132]   AT 01  : TaskName: CFW -- sys_timeout msecs=250,handler=0x880e5e21,arg=0x0
[22:06:55.132]          : Fn 1018421 T1 0768 T2 01 T3 02 Time 00 15 13 582
[22:06:55.134]   AT 01  : TaskName: Task Idle -- timer_sender handler=0x881645e1,arg=0x82139e70
[22:06:55.134]   AT 01  : TaskName: CFW -- sys_dotimeout handler=0x881645e1,args=0x82139e70
[22:06:55.234]   AT 01  : TaskName: CFW -- sys_timeout msecs=5000,handler=0x881645e1,arg=0x82139e70
[22:06:55.286]          : Fn 1018457 T1 0768 T2 11 T3 38 Time 00 15 13 746
[22:06:55.286]   AT 01  : TaskName: Task Idle -- timer_sender handler=0x880e5e21,arg=0x0
[22:06:55.286]   AT 01  : TaskName: CFW -- sys_dotimeout handler=0x880e5e21,args=0x0
[22:06:55.286]   AT 01  : TaskName: CFW -- tcp_slowtmr: processing active pcb
[22:06:55.286]   AT 01  : TaskName: CFW -- tcp_slowtmr: polling application
[22:06:55.286]   AT 01  : TaskName: CFW -- lwip_tcp_event conn=0x82124c38,lwip_event:4,pcb:0x8213a098
[22:06:55.286]   AT 01  : TaskName: CFW -- tcp_output: nothing to send (0x0)
[22:06:55.301]   AT 01  : TaskName: CFW -- tcp_output: snd_wnd 29312, cwnd 2848, wnd 2848, seg == NULL, ack 10480
[22:06:55.417]   AT 01  : TaskName: CFW -- sys_timeout msecs=250,handler=0x880e5e21,arg=0x0
[22:06:55.539]          : Fn 1018511 T1 0768 T2 13 T3 41 Time 00 15 13 996
[22:06:55.539]   AT 01  : TaskName: Task Idle -- timer_sender handler=0x880e5e21,arg=0x0
[22:06:55.539]   AT 01  : TaskName: CFW -- sys_dotimeout handler=0x880e5e21,args=0x0
[22:06:55.655]   AT 01  : TaskName: CFW -- sys_timeout msecs=250,handler=0x880e5e21,arg=0x0
[22:06:55.786]          : Fn 1018565 T1 0768 T2 15 T3 44 Time 00 15 14 247
[22:06:55.786]   AT 01  : TaskName: Task Idle -- timer_sender handler=0x880e5e21,arg=0x0
[22:06:55.786]   AT 01  : TaskName: CFW -- sys_dotimeout handler=0x880e5e21,args=0x0
[22:06:55.786]   AT 01  : TaskName: CFW -- tcp_slowtmr: processing active pcb
_**_[22:06:55.871]          : Detected event: 0x9db00000
[22:06:55.871]          : Detected event: 0x9db10000_**_
[22:06:55.971]   AT 01  : TaskName: CFW -- sys_timeout msecs=250,handler=0x880e5e21,arg=0x0

Neutree, 2017-12-27:
死机了么，在什么情况下出现的

lappaport, 2017-12-27:
跑mqtt demo的过程中，十五分钟内死机两次！

lappaport, 2017-12-27:
mqtt服务器是正常的，其他类型设备没问题。

Neutree, 2017-12-28:
改过代码么

lappaport, 2017-12-28:
只是改了mqtt的server ip，加入了mqtt 的user和password，其他没有动过。

lappaport, 2017-12-28:
#ifndef __DEMO_MQTT_H_
#define __DEMO_MQTT_H_

#define BROKER_IP  "119.23.29.210"
#define BROKER_PORT 1883
#define CLIENT_ID  "Neucrack"
#define CLIENT_USER  "mq5555tt"
#define CLIENT_PASS  "mq5555tt"
#define SUBSCRIBE_TOPIC "neucrack/gprs"
#define PUBLISH_TOPIC   "neucrack/app"
#define PUBLISH_INTERVAL 10000 //10s
#define PUBLISH_PAYLOEAD "hello I'm from gprs module"

#endif

    ci.client_id = CLIENT_ID;
    ci.client_user=CLIENT_USER;
    ci.client_pass=CLIENT_PASS;
    ci.keep_alive = 60;
    ci.clean_session = 1;
    ci.use_ssl = false;

其他未做任何改动。

Neutree, 2017-12-28:
我看看

Neutree, 2017-12-28:
连接失败，你的MQTT服务器没开

lappaport, 2017-12-28:
不好意思，这个是正确的地址，刚刚发到github上改了下的。

#define BROKER_IP  "119.23.209.214"
#define BROKER_PORT 1883
#define CLIENT_ID  "Neucrack"
#define CLIENT_USER  "mqtt"
#define CLIENT_PASS  "mqtt"
#define SUBSCRIBE_TOPIC "neucrack/gprs"
#define PUBLISH_TOPIC   "neucrack/app"
#define PUBLISH_INTERVAL 10000 //10s
#define PUBLISH_PAYLOEAD "hello I'm from gprs module"

非常抱歉，感谢。

lappaport, 2017-12-28:
麻烦用这个测试机。

Neutree, 2017-12-28:
### 宕机问题已经修复，获取最新代码即可。

造成死机是因为MQTT连接被断开的时候内存非法访问造成的，
造成断线的原因可能是网络不好，也有可能是有另一台机器使用同一个用户登录mqtt broker，导致broker迫使这个客户端断线

lappaport, 2017-12-28:
神速啊，厉害。

lappaport, 2017-12-28:
[image]
[image]
仍然在多次十几分钟内出现同样错误。1、目前只有一台a9的默认clientid登录。2、网络不好的情况可否sdk可否加入自动重连的机制。
感谢！

Neutree, 2017-12-28:
确认已经更新了么,下载这 [V1.4001](http://otge8q9ud.bkt.clouddn.com/GPRS_C_SDK_V1.401.7z)

lappaport, 2017-12-28:
是的，就是下午新下载的1.401重新编译，下载到板子上去的。

Joker-Zhao, 2018-01-04:
我使用代码版本V1.4001,未改过任何代码,直接编译烧写到A9开发板里面,也出现了这种问题

Neutree, 2018-01-15:
https://github.com/Ai-Thinker-Open/GPRS_C_SDK/releases/tag/V1.404

lappaport, 2018-01-16:
跑了12小时，mqtt正常，感谢。

closed 2018-01-16 by lappaport

Neutree, 2018-01-16:
这个问题后续我们还会继续跟踪的