Ai-Thinker-Open/GPRS_C_SDK #36: undefined reference to `sscanf'

brainbush, 2018-01-01:
```
LDGEN  by csdk    cust.ld

 
LD                app.elf
/cygdrive/d/projects/GPRS_C_SDK/build/app/app/lib/libapp_debug.a(cJSON.o): In function `print_number':
/cygdrive/d/projects/GPRS_C_SDK/app/src/cJSON.c:476: undefined reference to `sscanf'
value:0, p:ffffffff88242474, addend:0, symbol:0
/cygdrive/d/projects/GPRS_C_SDK/app/src/cJSON.c:476: relocation truncated to fit: R_MIPS16_26 against `sscanf'
make[2]: *** [/cygdrive/d/projects/GPRS_C_SDK/build/app//app.elf] Error 1
make[1]: *** [/cygdrive/d/projects/GPRS_C_SDK/build/app//app.srec] Error 2
make: *** [lod] Error 2

```

头文件中引用了stdio.h，在链接过程中报错。

brainbush, 2018-01-01:
还有，请问Trace()以及sprintf函数是不是都不支持%f的类型的输出？
谢谢

Neutree, 2018-01-09:
sscanf确实有问题，是的，暂时都不支持%f

Neutree, 2018-01-10:
暂时加入了一个sscanf，不过功能可能没有标准的全

closed 2018-01-10 by Neutree