Ai-Thinker-Open/GPRS_C_SDK #45: 关于 I2C 的疑问

brainbush, 2018-01-11:
1. 我们选用了一个基于I2C的温湿度传感器SHT30（[PDF手册见此](https://www.sensirion.com/fileadmin/user_upload/customers/sensirion/Dokumente/2_Humidity_Sensors/Sensirion_Humidity_Sensors_SHT3x_Datasheet_digital.pdf)），它的读取都是先写2字节长度的命令，随后直接读取（如图），而你们提供的I2C_ReadBytes()函数只能写1字节长的内容随后执行读取，虽可以用I2C_WriteBytes()来写2字节命令，但是无法单独调用读取来记录数据。请问能否提供单独拆开的读取部分或者可以指定 I2C_ReadBytes() 中的 memAddr 的长度之类的方式吗？谢谢！

[image]

2. https://github.com/Ai-Thinker-Open/GPRS_C_SDK/blob/51cd4caabc93e09be3e7a78248eb9be100256cd4/include/api_hal_i2c.h#L44
这里注释少了个参数（漏了个memAddr的说明）

谢谢！

Neutree, 2018-01-12:
额，好的，检查了一下I2C的API不是很好用，今天重新封装一下就好了

Neutree, 2018-01-12:
已经更新，可以获取最新代码尝试
```
git clone https://github.com/Ai-Thinker-Open/GPRS_C_SDK.git --recursive
```

brainbush, 2018-01-13:
嗯能用了，谢谢！

closed 2018-01-13 by brainbush