Ai-Thinker-Open/GPRS_C_SDK #56: cJSON依赖头文件

roamboy, 2018-01-20:
可否提供cJSON依赖的标准头文件
math.h, float.h, limits.h

roamboy, 2018-01-20:
报个cJSON链接问题
/cygdrive/d/Workspace/Src/A9G/gprs_c_sdk/build/gw/gw/lib/libgw_debug.a(gw_main.o): In function `print_number':
/cygdrive/d/Workspace/Src/A9G/gprs_c_sdk/gw/src/utils/cJSON/cJSON.c:502: undefined reference to `sscanf'
value:0, p:ffffffff8824135c, addend:0, symbol:0
/cygdrive/d/Workspace/Src/A9G/gprs_c_sdk/gw/src/utils/cJSON/cJSON.c:502: relocation truncated to fit: R_MIPS16_26 against `sscanf'
/cygdrive/d/Workspace/Src/A9G/gprs_c_sdk/build/gw/gw/lib/libgw_debug.a(gw_main.o): In function `parse_number':
/cygdrive/d/Workspace/Src/A9G/gprs_c_sdk/gw/src/utils/cJSON/cJSON.c:316: undefined reference to `strtod'
value:0, p:ffffffff882416e2, addend:0, symbol:0
/cygdrive/d/Workspace/Src/A9G/gprs_c_sdk/gw/src/utils/cJSON/cJSON.c:316: relocation truncated to fit: R_MIPS16_26 against `strtod'
make[2]: *** [/cygdrive/d/Workspace/Src/A9G/gprs_c_sdk/build/gw//gw.elf] Error 1
make[1]: *** [/cygdrive/d/Workspace/Src/A9G/gprs_c_sdk/build/gw//gw.srec] Error 2

punee995, 2018-01-21:
you can try **jsmn** lib which is minimalistic JSON parser in C.

roamboy, 2018-01-21:
it's only parser, hope support cJSON

Neutree, 2018-07-28:
cJSON was add to the lib with source code now

closed 2018-07-28 by Neutree