Ai-Thinker-Open/GPRS_C_SDK #57: 请问这是怎么回事

Destiny9701, 2018-01-20:
[image]

Destiny9701, 2018-01-20:
[image]
命令出错

Neutree, 2018-01-20:
解压好CSDTK3.8后，请不要使用记事本或者写字板编辑，
使用vscode ，notpad++，sublime text， atom等等这些编辑器都行

Neutree, 2018-01-20:
cygwin的结束符是`<LF>`即`\n`，记事本会把它变成`<CR><LF>`，即`\r\n`，这样做是不对的

closed 2018-01-20 by Neutree

Destiny9701, 2018-01-20:
@Neutree 我使用的是sublime text，这个有所了解

Destiny9701, 2018-01-20:
[image]
请问一下，那我重新下载再编辑？

Destiny9701, 2018-01-20:
好的,谢谢了，有可能是我双击的时候用记事本打开过什么了，我把这些重装试下。

Neutree, 2018-01-20:
把文件改成 <LF>换行就行了

Neutree, 2018-01-20:
http://blog.csdn.net/marco_90/article/details/42004109

http://www.codes51.com/itwd/4395205.html

Destiny9701, 2018-01-20:
好的，十分感谢解答，原来sublime text设置不对也会出现这样问题