Ai-Thinker-Open/GPRS_C_SDK #61: 最新的编译工具跟SDK，但是提示编译出错

itolddd, 2018-01-24:
$ build.sh demo mqtt
compile project mqtt
compile path /cygdrive/E/AiThinker/Code/GPRS_C_SDK_V1.406/demo/mqtt

MAKE              init
CC                sdk_init.c
      7 [main] ? (11724) E:\AiThinker\Tools\CSDTK\cygwin\crosscompiler\bin\mips-elf-gcc.exe: *** fatal error - couldn't allocate heap, Win32 error 487, base 0x4BF0000, top 0x4C00000, reserve_size 61440, allocsize 65536, page_const 4096
      7 [main] ? (11724) E:\AiThinker\Tools\CSDTK\cygwin\crosscompiler\bin\mips-elf-gcc.exe: *** fatal error - couldn't allocate heap, Win32 error 487, base 0x4BF0000, top 0x4C00000, reserve_size 61440, allocsize 65536, page_const 4096
      4 [main] mips-elf-gcc 7984 child_info::sync: wait failed, pid 11724, Win32 error 0
      4 [main] mips-elf-gcc 7984 child_info::sync: wait failed, pid 11724, Win32 error 0    477 [main] mips-elf-gcc 7984    477 [main] mips-elf-gcc 7984 fork: child -1 - died waiting for longjmp before initialization, retry 0, exit code 0x1000000, errno 11

请问这个是什么问题的？代码我一点都没改，Cygwin.bat是用管理员权限打开的。

Neutree, 2018-01-24:
这个版本cygwin本身的问题，重新启动cygwin，以后有时间优化一下编译环境

itolddd, 2018-01-24:
好的，我刚刚重新启动，能编译出来了。
问一个另外的问题，刚刚不能用work命令，是不是也是这个环境的问题？

Neutree, 2018-01-24:
work 是一个脚本，不能使用应该是你环境路径配置错了

itolddd, 2018-01-24:
我再研究下，谢谢

brainbush, 2018-01-24:
我之前也遇到过这样的问题，后来在其现有的cygwin目录下安装最新版本的cygwin就解决了，而且编译时间快了一半……
好像是由于cygwin的环境需要重新rebase才能有更大的内存空间导致的。
http://cygwin.wikia.com/wiki/Rebaseall

Neutree, 2018-01-24:
是的，更新了cygwin就好了

roamboy, 2018-01-25:
@brainbush 新版本再也不会出问题了么 ？

closed 2018-01-31 by Neutree