Ai-Thinker-Open/GPRS_C_SDK #64: A9G的SDK从LBS获得定位信息

Spring641, 2018-01-26:
请问SDK里面怎么读取LBS定位的数据呢？

Neutree, 2018-01-26:
现在还没有加入根据基站信息获取经纬度位置的功能，以后会加入。
现在如果着急用的话，可以获取基站信息用你自己的服务器或者网上免费的服务器做基站定位

Spring641, 2018-01-26:
因为目前用内置GPS定位没有信号，在空旷的地方测试过。

Neutree, 2018-01-26:
没信号的意思是gps能工作，有输出，但是插上gps天线后在空旷地方放置一段时间无法定位还是一颗星都搜不到呢。

检查天线是否正确
检查天线连接是否正确
检查gps是否能工作，通过开启gps后看是否有nmea信息输出辨别
检查供电能力是否能满足gps+gprs的供电要求
确定是在空旷区域能搜到卫星信号处测试

Spring641, 2018-01-26:
测试使用的是A9G开发板测试，使用A9Tracker测试，设置基站定位的时候可以显示模块的位置，在at指令设置使用gps定位时候，没有显示模块的位置。测试使用的电池是20C，600ma电池，电池电压4.1V。我天线连接正确，那有可能是测试地方搜不到卫星信号

Neutree, 2018-01-26:
使用at+gps=1  at+gpsrd=1看看有没有nmea信息输出

Spring641, 2018-01-26:
好的

Spring641, 2018-01-27:
[image]
在室内使用at+gps=1 at+gpsrd=1看看有没有nmea信息输出，有信息输出

Spring641, 2018-01-28:
已解决，换个大DB的天线就可以了

Destiny9701, 2018-02-14:
[image]
请问这个结构体成员各是什么？
[image]
与我所需要的信息有点对不上，能否解释下？

Neutree, 2018-06-29:
加入了LBS获取经纬度的方法，但是目前只支持国内，没有配置国外的数据库，如果有更高要求的需求，请自行搭建LBS服务器或者联系安信可升级服务

closed 2018-06-29 by Neutree