Ai-Thinker-Open/GPRS_C_SDK #70: MQTT error when sending large payload

ju-sti, 2018-01-30:
## 1. SDK version（SDK 版本）

{
V1.505

}

---

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）

{
When sending large MQTT payload, it shows MQTT error -1 (out of memory error). And crashes afterwards. Tested with mqtt payload = 20 (works), with payload = 270 or 642 it crashes.
[image]
In MQTT_Publish payloadLen is uint16 so payloads >255byte should be supported by SDK.

}

---

ju-sti, 2018-02-06:
@Neutree can you confirm this is a bug? thank you a lot

ju-sti, 2018-02-06:
can you confirm this is a bug, @Neutree , thank you

Neutree, 2018-06-29:
mqtt buffer length is 256, now changed to 4096 (Bytes), so topic length+payload length should not be more than 4096

closed 2018-06-29 by Neutree