Ai-Thinker-Open/GPRS_C_SDK #71: 关于api_event的问题

HelloAnpengyu, 2018-01-31:
## 1. SDK version（SDK 版本）

{
A9G
v1.506

}

---

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）

{

1.如何调用 TIME_GetRtcTIme(); 函数来获得RTC时间？
2.如何获得到基站定位的lac和cid？
3.API_EVENT_ID_SINGNAL_QUALITY 这个事件是自动触发的吗？如何手动得到csq呢？

}

---

Neutree, 2018-01-31:
直接调用函数就行啦

参考demo network2
目前是自动汇报

Neutree, 2018-01-31:
这个函数是获得本地的时间，如果没有设置，可能是不对的。
事件的时间是正确的时间，参数可以看api event . h
也可以调用TIME_SetIsAutoUpdateRtcTime设置为自动从网络更新本地时间

HelloAnpengyu, 2018-01-31:
事件里的事件怎么拿出来用呢。。
rtc = *(RTC_Time_t*)pEvent->param1;
我是这么写的，然后程序死掉了。。

Neutree, 2018-01-31:
请养成看注释的好习惯好么，别按照自己的思维理所当然

https://github.com/Ai-Thinker-Open/GPRS_C_SDK/blob/master/include/api_inc/api_event.h?L32

HelloAnpengyu, 2018-01-31:
还有就是基站定位的lac和lbs怎么拿到呢。。。

Neutree, 2018-01-31:
参考demo network2

前面就已经回答了

HelloAnpengyu, 2018-01-31:
好的，谢谢啦~~

closed 2018-05-14 by Neutree