Ai-Thinker-Open/GPRS_C_SDK #75: [main] sh 16612 _pinfo::dup_proc_pipe: DuplicateHandle failed, pid 16648, hProcess 0x324, wr_proc_pipe 0x364, Win32 error 5

itolddd, 2018-02-02:
## 1. SDK version（SDK 版本）

{

GPRS_C_SDK_V1505

}

---

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）

{

编译的时候总是会出现如下打印，然后就卡住了，需要重启Cygwin.bat才能解决。
编译前先clean也没用。

LODTOBIN  e:\AiThinker\Code\GPRS_C_SDK_V1.505\hex\app\app_flash.lod
LODTOBIN          Sucessful
LODCOMBINE        Combine user lod with Platform lod
LODCOMBINE        Combine sucessful
      1 [main] sh 16612 _pinfo::dup_proc_pipe: DuplicateHandle failed, pid 16648, hProcess 0x324, wr_proc_pipe 0x364, Win32 error 5
      1 [main] sh 16612 _pinfo::dup_proc_pipe: DuplicateHandle failed, pid 16648, hProcess 0x324, wr_proc_pipe 0x364, Win32 error 5

}

---

Neutree, 2018-02-02:
这应该是cygwin的问题，有能力的话可以自己升级cygwin就好

itolddd, 2018-02-03:
这个要怎么去升级？别人没有遇到吗？

Neutree, 2018-05-14:
更新到CSDTK4.2，不再使用cygwin，这个问题不会再出现了

closed 2018-05-14 by Neutree