Ai-Thinker-Open/GPRS_C_SDK #78: Trace(1,"%.*s",strlen(str), str); Not Working!

punee995, 2018-02-07:
v1.406

## 1. SDK version（SDK 版本）

{
**Not Working!**

`Trace(1,"%.*s",strlen(str), str); `

`sprintf(str2, "%.*s", strlen(str), str);`

}

---

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）

{

}

---

punee995, 2018-02-07:
also `char *strndup(const char *str, size_t size);` not supported
please add full support for **string.h**

Neutree, 2018-07-28:
* RE(Regular Expression) is not supported in sprintf or sscanf

* `strndup` not belongs to  standard of string.h, and it used malloc internally may make someone do some memory leak error, so you can add it by yourself, and you can event custom your `sprintf` or `sscanf` in `stdio.h`

closed 2018-07-28 by punee995