Ai-Thinker-Open/GPRS_C_SDK #79: Turn off Modem/GPS, Set MCU to sleep

ctandi, 2018-02-09:
Hi,

I'd like to build a GPS-Tracker based on the module. Is there any way to turn off the GSM-Modem and GPS-Receiver? I'd also like to put the MCU to timer based deep sleep. Is this possible?

Thanks

Neutree, 2018-02-10:
https://github.com/Ai-Thinker-Open/GPRS_C_SDK/blob/master/demo/pm/src/demo_pm.c

ctandi, 2018-02-11:
Thanks, but can you provide a brief explanation of what this function exactly does? Does it really power down the modem and the GNSS receiver? It seems like a lot of code besides the SDK is closed source. All I can find is binaries. Is there a way to get the full source code? My A9G is already on the way and I'd love to start developing as soon as it arrives. Thanks!