Ai-Thinker-Open/GPRS_C_SDK #80: How to Read string from sd card

punee995, 2018-02-10:
## 1. SDK version（SDK 版本）
{
v1.506
}
---
## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）
{

**1.       How to read string from sd card!!**

in demo_fs.c only write function( FsTFTest()  ) is given.
i'm trying to make function for read but it is not perfact so please help.

```
bool FsTFTestRd(char* stringData, char* fileName)
{
    int32_t fd;
    int32_t ret;
    bool retBool;
    uint8_t *unicodeName = NULL;
    uint32_t outlen = 0;
    retBool = LocalLanguage2UnicodeBigEndian(fileName, strlen(fileName)+1,CHARSET_UTF_8, &unicodeName, &outlen);
    if(!retBool)
        return false;  
    fd = API_FS_Open(unicodeName, (FS_O_RDONLY|FS_O_CREAT), 0);
    if ( fd < 0)
    {
        Trace(1,"Open file failed:%d",fd);
        OS_Free(unicodeName);
        return false;
    }
    ret = API_FS_Read(fd, (uint8_t*)stringData, strlen(stringData)) ;
    data_size = ret;
    Trace(1,"read ret:%d,strlen(stringData):%d",ret,strlen(stringData));
    API_FS_Close(fd);
    OS_Free(unicodeName);
    if(ret <= 0)
        return false;
    return true;
}
```
```
    char* readStr[255];
    char* file = "/t/config.txt";
    Trace(1,"Reading......");
    if(!FsTFTestRd(readStr,file))
        Trace(1,"Failed REading");
    Trace(1,"Done");
```

**2.    How can i ignore that Highlighted characters.**
[image]

}

---

Neutree, 2018-02-10:
memset stringData before read

Neutree, 2018-02-10:
memset stringData to zero before read

punee995, 2018-02-11:
Tried to  **memset stringData ** but it fails.

`memset(readStr,0,254);`

[image]

closed 2018-03-28 by punee995

Tanukaroy4494, 2018-08-10:
@punee995 How did you finally get rid of those highlighted characters?