Ai-Thinker-Open/GPRS_C_SDK #83: 浮点运算有关问题

Destiny9701, 2018-02-18:
## 1. SDK version（SDK 版本）

{

         V1.505
 

}

---

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）

{

math.h无法使用
aotf()报错
Trace不支持浮点
还有很多的参数没有注释，影响使用，有些注释不是很清晰

}

punee995, 2018-02-18:
please see issue #72.

Destiny9701, 2018-02-18:
thanks
不好意思，没看到这个类似的issue，但是有可能是我使用有错还是怎么回事，涉及浮点运算的就死机了，
例如这个操作
sprintf(testUart, "testA:%f", a);
Trace(2, "ready!!!");
Trace(2, "%s", testUart);
不知道为什么会死机，调试GDB不会使用，请问有文档吗？
万分感谢您在新年回复，祝您新年快乐

Destiny9701, 2018-02-18:
thanks
不好意思，没看到这个类似的issue，但是有可能是我使用有错还是怎么回事，涉及浮点运算的就死机了，
例如这个操作
char testUart[20];
sprintf(testUart, "testA:%f", a);
Trace(2, "ready!!!");
Trace(2, "%s", testUart);
不知道为什么会死机，调试GDB不会使用，请问有文档吗？
万分感谢您在新年回复，祝您新年快乐

closed 2018-02-19 by Destiny9701