Ai-Thinker-Open/GPRS_C_SDK #85: demo mqtt can't publish and scribe message

liunixgithub, 2018-02-24:
##1.现象
不做任何修改编译并加载demo mqtt即可复现问题
ubuntu中执行：
mosquitto_sub -h mqtt.neucrack.com -t "$neucrack/gprs" -v
无法收到消息
ubuntu中执行：
mosquitto_pub -h mqtt.neucrack.com -t "$neucrack/gprs" -m "asdfasdfq3212346"
模块也收不到消息
## 1. SDK version（SDK 版本）

{
V1.506
}
## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）

{
china mobile SIM card
demo socket worked well！
but demo mqtt can not work！ //不做任何修改编译并加载demo mqtt，无法工作
in ubuntu：
mosquitto_sub -h mqtt.neucrack.com -t "$neucrack/gprs" -v
no message received.
in ubuntu：
mosquitto_pub -h mqtt.neucrack.com -t "$neucrack/gprs" -m "asdfasdfq3212346"
no message received in board. //开发板也收不到消息
}

---

vencol, 2018-02-28:
开发板中的demodingyue订阅的topic是$neucrack/app，发布的topic才是$neucrack/gprs，而且他还有用户名和密码，而你的订阅和发布都没有

liunixgithub, 2018-02-28:
//服务器是支持匿名的，关掉匿名，添加用户
In server：mosquitto_sub -h m.5autos.cn -t $neucrack/gprs -u lwy -P lwylwy -v
//在服务器中执行上述命令订阅topic：$neucrack/gprs
In PC：mosquitto_pub -h m.5autos.cn -t $neucrack/gprs -u lwy -P lwylwy -m "asdf2223"
//在PC机上发布消息
server received “asdf2223” but couldn't receive any message from board
//服务器中可接收到“asdf2223”，但服务器无法接收到任何来自开发板的消息

vencol, 2018-03-01:
模块订阅的topic是#define SUBSCRIBE_TOPIC "$neucrack/app"，而你发送的topic是"$neucrack/gprs"

liunixgithub, 2018-03-01:
vencol你好，谢谢你的支持，我发送$neucrack/gprs这个topic的目的是测试我在服务器端的订阅是否可以收到消息，现在的现象是服务器可以收到pc端发送的消息，但收不到模块端发送的消息。另外模块也收不到订阅的$neucrack/app的消息，只不过我没有贴出来发消息的命令

SISYFirst, 2018-03-29:
你好，问题解决了吗，我也有同样的问题