Ai-Thinker-Open/GPRS_C_SDK #89: cygmin build路径问题

Littleoranga, 2018-03-07:
## 1. SDK version（SDK 版本）

{

}

---

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）

{

work路径没问题，ls也可以出现文件，但是运行./build.sh demo gpio就会报错，NO LIB FILE，找不到文件

}

---

vencol, 2018-03-07:
截个图看看，不太明白是什么问题

Littleoranga, 2018-03-07:
[image]

vencol, 2018-03-07:
你ls一下platform/csdk看下有没有debug，

vencol, 2018-03-07:
你ls一下platform/csdk看下有没有debug，看下有没有库，如果没有看下你是不是Git的时候没有

vencol, 2018-03-07:
你ls一下platform/csdk看下有没有debug，看下有没有库，如果没有看下你是不是Git的时候没有git clone https://github.com/Ai-Thinker-Open/GPRS_C_SDK.git --recursive，recursive是必须加上，才可以下库的，或者直接下载压缩包

Littleoranga, 2018-03-07:
我是直接下的压缩包，我ls了一下没有debug

vencol, 2018-03-07:
https://github.com/Ai-Thinker-Open/GPRS-C-SDK-LIB在这里下载下来，放到csdk里试试

Littleoranga, 2018-03-07:
我解压了放在CSTDK根目录下仍然不行

vencol, 2018-03-07:
不是根目录啊，是platform/csdk，提示是说这里没有debug

Littleoranga, 2018-03-07:
我找不到这个文件夹

vencol, 2018-03-07:
就在你ls那个截图都可以看到platform了，兄弟可以先自己找下吧

Littleoranga, 2018-03-07:
原来在工程目录下，好了问题解决了，多谢

vencol, 2018-03-08:
不客气