Ai-Thinker-Open/GPRS_C_SDK #94: 二次开发程序下载到2%断开

tcp01, 2018-03-09:
## 1. SDK version（SDK 版本）

{

GPRS_C_SDK_V1505

}

---

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）

{

    USB供电，每次下载到2%断开。

}

---

[image]
[image]

punee995, 2018-03-13:
**Try this ....**

https://github.com/Ai-Thinker-Open/GPRS_C_SDK/blob/master/doc/download_debug_tool_zh-cn.md#coolwatcher%E6%97%A0%E6%B3%95%E4%B8%8B%E8%BD%BD

Neutree, 2018-05-14:
重新连接试试，可以开始下载应该连接没有问题的，确认一下是不是电源提供能力不够

closed 2018-05-14 by Neutree