Ai-Thinker-Open/GPRS_C_SDK #102: Why LIS3DHx has been removed?

acubino, 2018-03-27:
I will receive my development board in next days, and i see that LIS3DHx is removed from it.

In the source code I can't see the acelerometer to. 

I need to use acelerometer in mi project, has been removed for a software issue (not ready) or something wrong with this ic?

updesh31, 2018-06-08:
Why removed I don't know. But accelerometer is working after soldered it. You can use i2c library without any problem.

Neutree, 2018-08-28:
nothing wrong with the IC, removed it just for the price!! you can solder it just like updesh31 do, or ask for the seller

mrgreywater, 2018-09-04:
Would be nice if there was maybe an version with accelerometer in the future. I would personally pay up to an extra $5 to not have to deal with soldering the small IC and potentially destroying the board in the process. Soldering it yourself is kinda difficult due to the size of the component. Also, an API for it would be nice as well.

dtekluva, 2019-04-24:
Hello please i need getting started with this board. i would really appreciate something as simple as an LED blinking application for trials.

punee995, 2019-04-24:
@dtekluva  
Same As #112 
First thing Create new Issue for off topic issue. (look for Closed Issue before opening new Issue).
Second, this is getting started guide [HERE](https://ai-thinker-open.github.io/GPRS_C_SDK_DOC/en/)

punee995, 2019-04-24:
@dtekluva please create a new issue if your problem  not related to "Why LIS3DHx has been removed?"
also, check out closed issue to find a quick solution.

harry91, 2020-03-14:
I am also willing to spend $ 5 more for LIS3DHx? Can you sell a version with LIS3DHx? I don't think anyone will compete with you on price.

breadcrumb3000, 2021-02-11:
Yes, I am also willing to spend $5 more for LIS3DHx!  Is there a demo or sample for using I2C with the LIS3DHx?