Ai-Thinker-Open/GPRS_C_SDK #105: 使用GPRS发送数据程序跑飞

colourfate, 2018-04-03:
## 1. SDK version（SDK 版本）

{
V1505
}

---

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）

{
我使用A9G模块，上面跑了两个线程，线程A计算量较大，以2Hz的速率运行，线程B开启GPRS，每5秒向服务器发送一次数据。在线程B发送第6次数据后，程序卡死，使用GDB查看报以下错误：
```
(gdb) bt f
#0  0x8200b8fd in sxr_Send ()
No symbol table info available.
#1  0x88091a45 in l1s_GetNBlockResult ()
No symbol table info available.
Backtrace stopped: frame did not save the PC
```
有时GDB进不去，这是少数几次成功进去后截取到的另一个保存：
```
#0  0x88095a2b in l1s_RxBcchJob ()
No symbol table info available.
#1  0x81c01f40 in ?? ()
No symbol table info available.
```
**1. 单独运行线程A和线程B均正常，同时运行就会报错。**
**2. 从报错信息看出错位置似乎不在我写的程序中。**
请问这是SDK的问题吗？是的话如何解决？
}

---

mymx, 2018-04-03:
同学，留个联系方式吧，我和你一个目的的

colourfate, 2018-04-04:
@mymx chdlogitech@163.com

mymx, 2018-04-04:
@colourfate 我通过邮箱给你发个qq