Ai-Thinker-Open/GPRS_C_SDK #110: See carrier network name

acubino, 2018-04-11:
I want to know the carrier name of the connected network. The posibility of see the carrier network name can be very interesting (for example choose the correct APN if we use different cards).

In the current SDK is not possible to read that info, like "Vodafone", "China Mobile" ...

punee995, 2018-04-26:
You can not get "Vodafone" or "China Mobile" right out of A9G but you can use MCC and MNC to determine  Network Provider and Country As well.
see this  [network2 demo](https://github.com/Ai-Thinker-Open/GPRS_C_SDK/tree/master/demo/network2)

acubino, 2018-04-26:
Using the MCC and MNC you can know who is the owner of the network, if you are not in roaming (including fake local roaming) the owner and you operator is the same, but in roaming you need to use the network name in order to know who is your operator.

punee995, 2018-04-26:
you can also get your carrier name from **ICCID** [see here](https://www.emnify.com/2016/05/06/emnify-8988303-iccid-iin-and-sim-serial-number-explained/).
For Testing (8991981798993312333 - airtel).
and from **IMSI**(don't know how to get this in A9g) as it contain MCC and MNC. 
FYI IMSI No. and ICCID No. does not change if you're roaming.

acubino, 2018-04-26:
I was explore that option, but almost un Spain can't be valid, because some virtual operators have the same issuer identifier in ICCID but not the same APN.
I check how Android and Microsoft do it, and in most case they use all params to choose. Sometimes use the ICCID or some no.

You can check the Android XML here.
https://github.com/LineageOS/android_vendor_cm/blob/cm-14.1/prebuilt/common/etc/apns-conf.xml

Microsoft use COSA / APN database, but requires more work due use different XML files.
https://docs.microsoft.com/en-us/windows-hardware/drivers/mobilebroadband/cosa-apn-database

The problem is that in both cases requires the Carrier name. I don't know if some CSDK functions can show this info, but there isn't any doc about CSDK and I can't do directly.

punee995, 2018-04-26:
right now i'm using SMS to setup APN name it's easy way but auto-apn is best.

```
    if(strcmp(msg[0],"apn") == 0){
        if (msg_len == 4){
        //#apn#airtelgprs.com#uname#pwd
            data.apnName = msg[1];
            data.apnUser = msg[2];
            data.apnPass = msg[3];
            SendSMS(hdr[0],"APN setup Ok.");
        }
        else if (msg_len == 3){
        //#apn#airtelgprs.com#uname
            data.apnName = msg[1];
            data.apnUser = msg[2];
            data.apnPass = '\0';
            SendSMS(hdr[0],"APN setup Ok.");
        }
        else if (msg_len == 2){
        //#apn#airtelgprs.com
            data.apnName = msg[1];
            data.apnUser = '\0';
            data.apnPass = '\0';
            SendSMS(hdr[0],"APN setup Ok.");
        }
```

acubino, 2018-04-26:
Use SMS is one option, but it require SMS cost and user action. When you are making a new design you will try that must be the easiest and more automate possible.

mnorval, 2019-08-27:
Does anyone know how I can extract the mobile number/phone number?