Ai-Thinker-Open/GPRS_C_SDK #122: A9g board programming

ShahzadHamid, 2018-05-06:
Please someone guide me can I programme my a9g board with Arduino IDE software?  I am much confused about that how to program my module. Please help I am new on this module. 
Thanks in Advance

capantoja, 2018-05-12:
I understand that the programming from the arduino ide is not yet supported, to program the module an app called coolwatcher is used and the pins HST_TX and HST_RX are used through a USB - Serial converter, to see the complete procedure follow the instructions of the video
https://www.bilibili.com/video/av16579395/

Neutree, 2018-05-14:
* documentation [here](https://ai-thinker-open.github.io/GPRS_C_SDK_DOC/en/), and 'how to intall' [here](https://ai-thinker-open.github.io/GPRS_C_SDK_DOC/en/c-sdk/installation.html)
* no arduino IDE support, download firmware with coolwatcher(RDA provided)

closed 2018-05-29 by Neutree

jumpjack, 2018-10-22:
news about IDE support?

jumpjack, 2019-07-15:
up!

majidmp97, 2020-12-03:
**is possible program a9g directly without microcontroller?**