Ai-Thinker-Open/GPRS_C_SDK #125: gps demo not working in A9G-dev board

capantoja, 2018-05-12:
## 1. SDK version（SDK 版本）

{

SDK V1505

}

---

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）

{

I compile and load the gps demo an watch the trace window
[image]

the latitude and longitude is empty.

}

---

punee995, 2018-05-13:
please check demo with **Active antenna** if available if not then try it in **open area**.

hjf, 2018-06-11:
I also have this problem. I'm using what I believe is an active antenna (it's thick and heavy). I enabled other traces in the code but I don't think I'm even getting the NMEA data in UART1.
I'm not in open area but the antena is out in the window with view of the sky.

hjf, 2018-06-11:
Actually the problem was something else. The demo needs a SIM card connected:

```
    switch(pEvent->id)
    {
        case API_EVENT_ID_GPS_UART_RECEIVED:
        //     Trace(1,"received GPS data,length:%d, data:%s,flag:%d",pEvent->param1,pEvent->pParam1,flag);
            if(flag)
            {
                Buffer_Puts(&gpsNmeaBuffer,pEvent->pParam1,pEvent->param1);
                GpsUpdate();
            }
            break;
        case API_EVENT_ID_UART_RECEIVED:
            if(pEvent->param1 == UART1)
            {
                uint8_t data[pEvent->param2+1];
                data[pEvent->param2] = 0;
                memcpy(data,pEvent->pParam1,pEvent->param2);
                Trace(1,"uart received data,length:%d,data:%s",pEvent->param2,data);
            }
            break;
        case API_EVENT_ID_NETWORK_REGISTERED_HOME:
        case API_EVENT_ID_NETWORK_REGISTERED_ROAMING:
            Trace(1,"register success");
            flag = 1;
            break;
        default:
            break;
    }
``` 

Without a SIM, the variable flag is never set.

closed 2018-06-29 by Neutree