Ai-Thinker-Open/GPRS_C_SDK #130: How to get minute from time.timeZone

punee995, 2018-05-22:
## 1. SDK version（SDK 版本）

{

      V1.506

}

---

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）

{

Tried demo/time but it displays only +5 hour but it need to be +5:30 in india.
How can i get timeZone minute???

[image]

}

---

Neutree, 2018-05-28:
https://github.com/Ai-Thinker-Open/GPRS_C_SDK/releases/tag/V2.100

Neutree, 2018-05-28:
https://github.com/Ai-Thinker-Open/GPRS_C_SDK/releases/tag/V2.100

https://github.com/Ai-Thinker-Open/GPRS_C_SDK/blob/master/include/api_inc/api_inc_time.h#L44

closed 2018-05-28 by punee995