Ai-Thinker-Open/GPRS_C_SDK #132: Where do I get the right version of milps-elf-gcc or what is the right configuration to build it with so that it may work with the SDK?

cryarchy, 2018-05-27:
## 1. SDK version（SDK 版本）

{
V2.001
}

---

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）

{
Operation: Building any application.
To Reproduce: In the folder *GPRS_C_SDK_V2.001* run `bash build.sh app`
}

---

Command output:
```
param number:1
compileMode:debug
compile project app
compile path /home/archy/A9G/GPRS_C_SDK_V2.001/app
core number:4
Build host is LINUX

MAKE              init
CC                sdk_init.c
mips-elf-gcc: error: unrecognized argument in option '-march=xcpu'
mips-elf-gcc: note: valid arguments to '-march=' are: 10000 1004kc 1004kf 1004kf1_1 1004kf2_1 10k 12000 12k 14000 14k 16000 16k 2000 20kc 24kc 24kec 24kef 24kef1_1 24kef2_1 24kefx 24kex 24kf 24kf1_1 24kf2_1 24kfx 24kx 2k 3000 34kc 34kf 34kf1_1 34kf2_1 34kfx 34kn 34kx 3900 3k 4000 4100 4111 4120 4130 4300 4400 4600 4650 4700 4k 4kc 4kec 4kem 4kep 4km 4kp 4ksc 4ksd 5000 5400 5500 5900 5k 5kc 5kf 6000 6k 7000 74kc 74kf 74kf1_1 74kf2_1 74kf3_2 74kfx 74kx 7k 8000 8k 9000 9k from-abi i6400 interaptiv loongson2e loongson2f loongson3a m14k m14kc m14ke m14kec m4k m5100 m5101 mips1 mips2 mips3 mips32 mips32r2 mips32r3 mips32r5 mips32r6 mips4 mips64 mips64r2 mips64r3 mips64r5 mips64r6 native octeon octeon+ octeon2 octeon3 orion p5600 r10000 r1004kc r1004kf r1004kf1_1 r1004kf2_1 r10k r12000 r12k r14000 r14k r16000 r16k r2000 r20kc r24kc r24kec r24kef r24kef1_1 r24kef2_1 r24kefx r24kex r24kf r24kf1_1 r24kf2_1 r24kfx r24kx r2k r3000 r34kc r34kf r34kf1_1 r34kf2_1 r34kfx r34kn r34kx r3900 r3k r4000 r4100 r4111 r4120 r4130 r4300 r4400 r4600 r4650 r4700 r4k r4kc r4kec r4kem r4kep r4km r4kp r4ksc r4ksd r5000 r5400 r5500 r5900 r5k r5kc r5kf r6000 r6k r7000 r74kc r74kf r74kf1_1 r74kf2_1 r74kf3_2 r74kfx r74kx r7k r8000 r8k r9000 r9k rm7000 rm7k rm9000 rm9k sb1 sb1a sr71000 sr71k vr4100 vr4111 vr4120 vr4130 vr4300 vr5000 vr5400 vr5500 vr5k xlp xlr
mips-elf-gcc: error: unrecognized argument in option '-mtune=xcpu'
mips-elf-gcc: note: valid arguments to '-mtune=' are: 10000 1004kc 1004kf 1004kf1_1 1004kf2_1 10k 12000 12k 14000 14k 16000 16k 2000 20kc 24kc 24kec 24kef 24kef1_1 24kef2_1 24kefx 24kex 24kf 24kf1_1 24kf2_1 24kfx 24kx 2k 3000 34kc 34kf 34kf1_1 34kf2_1 34kfx 34kn 34kx 3900 3k 4000 4100 4111 4120 4130 4300 4400 4600 4650 4700 4k 4kc 4kec 4kem 4kep 4km 4kp 4ksc 4ksd 5000 5400 5500 5900 5k 5kc 5kf 6000 6k 7000 74kc 74kf 74kf1_1 74kf2_1 74kf3_2 74kfx 74kx 7k 8000 8k 9000 9k from-abi i6400 interaptiv loongson2e loongson2f loongson3a m14k m14kc m14ke m14kec m4k m5100 m5101 mips1 mips2 mips3 mips32 mips32r2 mips32r3 mips32r5 mips32r6 mips4 mips64 mips64r2 mips64r3 mips64r5 mips64r6 native octeon octeon+ octeon2 octeon3 orion p5600 r10000 r1004kc r1004kf r1004kf1_1 r1004kf2_1 r10k r12000 r12k r14000 r14k r16000 r16k r2000 r20kc r24kc r24kec r24kef r24kef1_1 r24kef2_1 r24kefx r24kex r24kf r24kf1_1 r24kf2_1 r24kfx r24kx r2k r3000 r34kc r34kf r34kf1_1 r34kf2_1 r34kfx r34kn r34kx r3900 r3k r4000 r4100 r4111 r4120 r4130 r4300 r4400 r4600 r4650 r4700 r4k r4kc r4kec r4kem r4kep r4km r4kp r4ksc r4ksd r5000 r5400 r5500 r5900 r5k r5kc r5kf r6000 r6k r7000 r74kc r74kf r74kf1_1 r74kf2_1 r74kf3_2 r74kfx r74kx r7k r8000 r8k r9000 r9k rm7000 rm7k rm9000 rm9k sb1 sb1a sr71000 sr71k vr4100 vr4111 vr4120 vr4130 vr4300 vr5000 vr5400 vr5500 vr5k xlp xlr
mips-elf-gcc: error: unrecognized command line option '-mmips-tfile'
make[3]: *** [/home/archy/A9G/GPRS_C_SDK_V2.001/build/app/init/obj/debug/sdk_init.o] Error 1
make[3]: *** Waiting for unfinished jobs....
make[2]: *** [dependencies] Error 2
make[1]: *** [/home/archy/A9G/GPRS_C_SDK_V2.001/build/app//app.srec] Error 2
make: *** [lod] Error 2
=== Build Time: 0s at 2018-05-27 19:28:05 ===
!!!!!!!!!!!!!!!!!!!!
   BUILD FAILED
!!!!!!!!!!!!!!!!!!!!
```

I have tried building multiple configurations of **mips-elf-gcc** on my **Linux Mint 17.3 Cinnamon 64-bit** to no avail. Where can I get a prebuilt version that will work with the SDK or what are the correct dependency versions and configurations to get it working with the SDK?

Neutree, 2018-05-28:
linux not supported yet

cryarchy, 2018-05-28:
Thanks, @Neutree ! Shifting to windows.

closed 2018-05-28 by cryarchy