Ai-Thinker-Open/GPRS_C_SDK #135: math function atan2()

updesh31, 2018-05-30:
## 1. SDK version（SDK 版本）

{

    SW_V2101_csdk

}

---

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）

{

	uint8_t tog = 0;
	int16_t Buffer[3];
	double x_Buff;
	double y_Buff;
	double z_Buff;
	float rolldiff = 0.0,pitchdiff =0.0;
	double roll = 0.00,preroll=0.0, pitch = 0.00,prepitch = 0.0;
    I2C_Config_t config;
    config.freq = I2C_FREQ_100K;
    I2C_Init(I2C_ACC, config);
	tog = LIS3DSH_ReadID();
	if(tog == 0x33)
	{
	  LIS3DSH_Initalize();
	  sprintf(temp,"Accelerometer Initalized\r\n");
	  UART_Write(UART1,temp,strlen(temp));
	}
 while(1)
    {
		Buffer[0] = 0;
		Buffer[1] = 0;
		Buffer[2] = 0;
		LIS3DSH_ReadACC(Buffer);
		x_Buff = Buffer[0];
		y_Buff = Buffer[1];
		z_Buff = Buffer[2];
		if(x_Buff == 0 && y_Buff == 0)
		{
			LIS3DSH_RebootCmd();
			LIS3DSH_Initalize();
		}
		roll = atan2(y_Buff , z_Buff) * 57.3;
		pitch = atan2((- x_Buff) , sqrt(y_Buff * y_Buff + z_Buff * z_Buff)) * 57.3;
              OS_Sleep(100);
}

}

---

Neutree, 2018-06-02:
https://github.com/Ai-Thinker-Open/GPRS_C_SDK/releases

updesh31, 2018-06-02:
Thankyou for your quick solution.
:)

updesh31, 2018-06-03:
but same compiler issue with sqrt() function? Actually I am working on GPS tracker with A9G. Are all these events functioning "API_EVENT_ID_NETWORK_REGISTER_SEARCHING,API_EVENT_ID_NETWORK_REGISTER_DENIED,API_EVENT_ID_NETWORK_REGISTER_NO"? Because after 2 hours module closes the connection with server and never starts again.

punee995, 2018-06-03:
For 2nd answer you need to check for A9g is connected to inernet or not after some time.

HINT:-

API_EVENT_ID_NETWORK_DETACHED,
API_EVENT_ID_NETWORK_ATTACH_FAILED,
API_EVENT_ID_NETWORK_DEACTIVED,
API_EVENT_ID_NETWORK_ACTIVATE_FAILED

updesh31, 2018-06-04:
I connected a winbond flash memory with SPI interface but it's not working.Any suggestion. I followed the example but not able to success.

punee995, 2018-06-04:
Need detailed info.!

Please create new issue for this.

updesh31, 2018-06-04:
Now GPS tracker is working perfectly.After few days testing I will update the result.
Thanks

updesh31, 2018-06-26:
Compiler error with sqrt()
undefined reference to `sqrt' it's the same error that was with atan2() before new release sdk.

closed 2018-07-31 by updesh31