Ai-Thinker-Open/GPRS_C_SDK #136: UART_READ接收不到串口数据

sqm289798469, 2018-06-04:
## 1. SDK version（SDK 版本）

{

2.102

}

---

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）

{
Arduino TX,RX连接A9G pubing板子的UART1，API_EVENT_ID_UART_RECEIVED事件里用UART_READ接收不到Arduino发过来的数据

}

---

punee995, 2018-06-04:
There's two method to receive uart data

1. using event

UART_Config_t config = {
        .baudRate = UART_BAUD_RATE_115200,
        .dataBits = UART_DATA_BITS_8,
        .stopBits = UART_STOP_BITS_1,
        .parity   = UART_PARITY_NONE,
        .rxCallback = NULL,
        .useEvent = true,
    };

2 using call back function

UART_Config_t config = {
        .baudRate = UART_BAUD_RATE_115200,
        .dataBits = UART_DATA_BITS_8,
        .stopBits = UART_STOP_BITS_1,
        .parity   = UART_PARITY_NONE,
        .rxCallback = myCallbackFunction,
        .useEvent = false,
    };

Try demo/uart and short RX1 and TX1 pin together

SISYFirst, 2018-10-30:
UART_Config_t config = {
			        .baudRate = UART_BAUD_RATE_115200,
			        .dataBits = UART_DATA_BITS_8,
			        .stopBits = UART_STOP_BITS_1,
			        .parity   = UART_PARITY_NONE,
			        .rxCallback = OnUart1ReceivedData,
			        .useEvent = false,
  			  };
		UART_Init(UART1,config);
		uint8_t temp[20];
       		 uint8_t buffer[50];
		
	        sprintf(temp,"hello:%d,1234567890 1234567890\n", ++times);
        	UART_Write(UART1,temp,strlen(temp)+1);

void OnUart1ReceivedData(UART_Callback_Param_t param)
{
    memset(rxbuffer,0,sizeof(rxbuffer));
    uint32_t len = UART_Read(UART1,rxbuffer,param.length,200);
    Trace(1,"uart1 received data,length:%d,read:%d,data:%s",param.length,len,rxbuffer);
}
uart1 received data,length:33,read:-2011735367,data: what's the problem?????

SISYFirst, 2018-10-30:
RX1 and TX1 pin are connected to together yet

vbtaneja, 2019-01-26:
temp is allocated a space of 20 but the print string is much longer.
Try temp[50] instead

fogbox, 2019-03-06:
Please, at least add possibility to change timeout between bytes for UART1. I have an dumb device wich freeze byte stream just for about 10 ms in one packet transfer. As result - OS twice call into callback function while just one incomming packet transfer. I can cobmine data and check time for determine end of transfer packet, but can`t come up with how do this pretty and simple.
And.. thank`s for your work! :) It`s really cool device.