Ai-Thinker-Open/GPRS_C_SDK #137: SPI flash memory

updesh31, 2018-06-04:
## 1. SDK version（SDK 版本）

{

SW_V2102_csdk

}

---

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）

{
I connected a winbond flash memory with SPI interface but it's not working.Any suggestion. I followed the example but not able to success.

MISO - PIN-4
MOSI - PIN-5
CS0  - PIN-7
CLK - PIN-8

There is no FirstBit (MSB/LSB) selection.
    SPI_Config_t SPI_CFG =
    {
	 .cs = SPI_CS_0,
        .txMode = SPI_MODE_DIRECT_POLLING,
        .rxMode = SPI_MODE_DIRECT_POLLING,
        .freq = 3000000,
        .line = SPI_LINE_3,
        .txOnly = false,
        .cpol = 0,
        .cpha = 0,
        .csActiveLow = true,
        .dataBits = SPI_DATA_BITS_8,
        .irqHandler = NULL,
        .irqMask = {0,0,0,0,0}
    };

    SPI_Init(SPI1, SPI_CFG);

}

---

updesh31, 2018-07-28:
Can I use flash memory with SPI?

updesh31, 2018-08-07:
I tried a lot but not getting success.

updesh31, 2018-08-13:
here is the output of A9G as tracker without data saving.....(I need SPI function.)
[image]

Neutree, 2018-08-25:
You may used the SPI lib with wrong way, check your code again and debug with logic analyzer

closed 2018-09-01 by updesh31

MaBarakat, 2024-05-15:
@updesh31 are you find the solution?

MaBarakat, 2024-08-07:
@Neutree 
hi,你好
我想問的是在A9G模組中使用spi寫入快閃記憶體還是必須使用TF卡 ?
請幫忙並解答

I am asking about using spi to write on flash memory or must using TF card in A9G module ?
Please help and answer

Neutree, 2024-08-07:
@MaBarakat https://ai-thinker-open.github.io/GPRS_C_SDK_DOC/en/c-sdk/function-api/file-system.html

MaBarakat, 2024-08-07:
@Neutree 
抱歉，答案是否定的，它需要 TF 卡

I'm sorry, but the answer is no, it requires a TF card

Neutree, 2024-08-07:
[image]
@MaBarakat 

https://github.com/Ai-Thinker-Open/GPRS_C_SDK/blob/master/demo/fs/src/demo_fs_new.c
[image]

Neutree, 2024-08-07:
@MaBarakat  https://ai-thinker-open.github.io/GPRS_C_SDK_DOC/zh/c-sdk/gong-neng-ji-api/fswen-jian-xi-7edf2928-chu-cun-shu-ju-dao-flash-huo-t-536129.html see Chinese doc.

MaBarakat, 2024-08-07:
@Neutree 
我指的是外部閃存，例如 W25Q 閃存
I mean on external flash memory like W25Q FLASH Memory

ErickXHdez, 2024-09-14:
@MaBarakat are you find some solution?

MaBarakat, 2024-09-14:
@ErickXHdez No until now.

ErickXHdez, 2024-09-14:
Ok i can write but still i can read anything @Neutree @punee995 @MaBarakat

MaBarakat, 2024-09-14:
@ErickXHdez how you know  writing success if you don't read it

ErickXHdez, 2024-09-14:
Because i have in the other part a esp32 like a spi slave

MaBarakat, 2024-09-14:
The way to write on Microcontroller differs from spi flash to sure the writing success try to read from it by other Microcontroller not A9G.