Ai-Thinker-Open/GPRS_C_SDK #140: Unable to download any on pundit a9g

bitwarriors, 2018-06-12:
Goodmorning,
after some little problems on making the lod files, now (around 2 weeks) I'm not able to download the new firmware. My coolwatcher (2.00.02) stops on this error, after a workig first connection: 

`CHST_Read32': Request to coolhost timed out ! (CoolHostTimeoutError)

./rbbase/common/coolhost_connections.rb 161 `CHST_Read32'
./rbbase/common/coolhost_connections.rb 161 `read32'
./rbbase/common/base.rb  85   `read'
./rbbase/common/base.rb  239  `read'
./chipgen/Modem2G/toolpool/scripts/8809/target_executor.rb 72  `enterHostMonitor'
./chipgen/Modem2G/toolpool/scripts/ChipStd/target_executor.rb  366  `session'
./chipgen/Modem2G/toolpool/scripts/ChipStd/ramrun.rb  12  `doramrunXfer'
./chipgen/Modem2G/toolpool/scripts/ChipStd/ramrun.rb  62  `ramrunXfer'
./chipgen/Modem2G/toolpool/plugins/fastpf/fastpf.rb   212 `dofastpfXfer'
./chipgen/Modem2G/toolpool/plugins/fastpf/fastpf.rb   718 `dofastpf'
./chipgen/Modem2G/toolpool/plugins/fastpf/fastpf.rb   788 `fastpf'
./chipgen/Modem2G/toolpool/plugins/fastpf/fastpfplugin.rb  83  `fastpfGenericGo'
./chipgen/Modem2G/toolpool/plugins/fastpf/fastpfplugin.rb  112  `fastpfGo'

the CSDK is the 4.2 and the GPRS_SDK is the latest on line, the usb-uart is the cp2102 and the tx->hst-rx and rx->hst-tx pins are connected.
I'm very frustrated because no way to get rid of this error.
Any helps will be gratefully accepted 👍 
Thank you

Neutree, 2018-06-12:
refer to here:https://ai-thinker-open.github.io/GPRS_C_SDK_DOC/en/c-sdk/burn-debug.html#coolwatcher-download-fail

bitwarriors, 2018-06-13:
Thank you Neutree for your quick reply, unfortunately before posting my request I've already followed the hints on that place :-( . I don't understand if is a device problem or a connection problem. When I fire-up the coolhost application do I have to read more than the only Open Com4 etc etc ?
Thank You again

lamqhoang, 2018-06-14:
Hi @bitwarriors ,
Make sure you have the good wiring, I had the same issue with a long jumper wire before. Even if you have connected the USB-UART and the A9G board to the same computer, it is recommended to have a jumper wire for GND.

Neutree, 2018-06-14:
you can try [AT firmware upgrade tool](http://wiki.ai-thinker.com/_media/ai-thinker_gsm_module_download_tool_v11.00.04.7z) first, and [AT firmware here](http://wiki.ai-thinker.com/_media/gprs/a9g-v02.01.20180416rc.zip),
[image],  uart COM port(s) will appear if hardware connection is ok,

bitwarriors, 2018-06-15:
Thanks all my friends,
sadly i can't test because I'm not in my place, buut as soon as I get home I 'll check !
I will keep you informed on my progresses :-))

closed 2018-06-29 by Neutree

bitwarriors, 2018-07-03:
Hi all!
I tried all the solutions you told me and seems there is a problem about the connection, means the usb uart cp2102 is connected and I have the evidence in the hardware list 
[image]
but when i open any of the sever tool the result is I can't use the device 
[image]

AAARGH!

bitwarriors, 2018-07-03:
Ok ... Miracle! I don't know why but now all is working !
I uploaded a demo / gps ... but now, how can i monitor if the gps is working trhough a terminal, like arduino method?
thank you all

dhernandezl, 2019-12-17:
Did you manage to solve permanently?
I have the same problem

bitwarriors, 2019-12-17:
Hi … not permanently … infact I don’t understand where is the real problem … can we submit again tighter the request ?

Tad3a, 2023-07-20:
Did you find any solution?

bitwarriors, 2023-07-20:
Not yet

Tad3a, 2023-07-20:
Is this good in  coolwatcher?
[image]
But when I try to launch fastpf then I get this familiar error :D
[image]
I use this CP2102 TTL convertor: https://www.laskakit.cz/prevodnik-6pin-usb-ttl-uart--cp2102--dtr-pin/
Do you think it's the TTL converter's fault?

collins-kariuki, 2024-04-07:
@tadeastehan any breakthrough on this?

tadeastehan, 2024-04-07:
No, I gave up trying to get it to work.

collins-kariuki, 2024-04-07:
Alright.
I'm still digging around.
Thanks for the prompt response.

tadeastehan, 2024-04-07:
If you find the solution to this problem, let me know.

MCPM23, 2024-04-27:
[image]
I fixed it by setting the port beforehand in this window. The problem now, after flasing it, it's that the module doesn't turn on :( so I'm having again this issue

BSPATELSB, 2024-06-25:
Hello Have resolved this problem, by applying below steps...

-> First verify the GND connection, If you give the power from USB micro cable then do not connect GND wire
-> after this validation, restart coolwatcher application 
-> Verify your COM port carefully
-> Then after COM port connection confirmation, flash both .lod files
-> You should find that, the flash error has resolved

26mrlon, 2024-09-28:
`CHST_Read32': Request to coolhost timed out! (CoolHostTimeoutError)

Hello, guys! I found a solution to this error because I encountered it too. It took me almost two days to figure it out, but I finally solved it today.

First, you need to connect the RX pin to the HST_TX and the TX pin to the HST_RX.

Next, take your micro USB cable, connect it to your computer, and then to your A9G board. This means the power is coming from the micro USB.

Before opening CoolWatcher, press the PWR button on the A9G board.

Then, flash the two LOD files.