Ai-Thinker-Open/GPRS_C_SDK #141: Is it possible to read and write internal variables with SDK ?

gsmwhatnot, 2018-06-15:
## 1. SDK version（SDK 版本）

{

4.2

}

---

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）

{

I am trying to read and write some variables inside module, for example read kc_low and kc_high or TMSI, is it possible to achieve this goal by calling their pointers ? or is it possible to call functions in STACK part inside SDK ? For example directly calling 'mmc_ImsiDetach' inside SDK

}

---

Neutree, 2018-06-16:
it wont be supported cause we think most of people don't need to know how gprs reigster works(communication or encryption)  when they are working on their IOT projects.

And if you want to do more, maybe use the source code from RDA more proper

gsmwhatnot, 2018-06-17:
Thanks Neu for your reply, I tried compiling RDA8955 source code but when I flashed LOD file with coolwatcher into the A9 module it bricked the module and it is not responding to any command ( AT firmware also failed to flash ) Is there any source code for your AT firmware ? I couldn't find it on ai-thinker.com

Neutree, 2018-06-17:
ai-thinker do not provide any source code of at firmware but you can find it by other ways

gsmwhatnot, 2018-06-17:
Thank you again, I tried different sources but compiled LOD is not working and causing modules became non-responsive. Right now I have like 10 dead A9 modules.
 I have few questions it would be great if I get some answers : 
1. Is there any paid support from AI-Thinker ?
2. Is it possible to buy AT firmware from you ?
3. What are the ways for contacting AI-Thinker ? My company tried phone and email without any success
4. Is CZDxxx666 at gmail your official email address ? ( those xxx are for hiding your complete email address )
5. Is it possible for AI-Thinker to build custom A9 firmware ?

Thank you

Neutree, 2018-06-17:
most at firmware source code from rda use UART2 by default, you can try to send at command though UART2 after you download lod.

Neutree, 2018-06-17:
the google email is personal....

gsmwhatnot, 2018-06-17:
no success on UART2
By 'non-responsive' I mean even Host UART ( HRx, HTx ) has no response in the coolwatcher. It seems that I am missing some bootloader parts or make variables !
I am using this command to build the firmware : 
`ctmake CT_TARGET=8955_modem CT_RELEASE=debug CT_USER=FAE CT_MODEM=1`

gsmwhatnot, 2018-06-17:
My company tried both corporate email addresses but there were no response from them. 
You should have received an email from info at terarange

gsmwhatnot, 2018-06-18:
Thanks Neu, I hope we get response soon from AI-Thinker.
Guys, please excuse me for using this discussion in not so-professional way.

closed 2018-06-18 by gsmwhatnot

samigroup, 2023-03-25:
> Thank you again, I tried different sources but compiled LOD is not working and causing modules became non-responsive. Right now I have like 10 dead A9 modules. I have few questions it would be great [...]

Hello, 
Did you get this worked out? I was able to recover mine.  I can try to do it for your parts. 

Sami