Ai-Thinker-Open/GPRS_C_SDK #146: undefined reference to 'strtod'

cryarchy, 2018-06-22:
## 1. SDK version（SDK 版本）
{
2.001
}

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）
{
Converting a string to a double:

To reproduce use `strtod` to convert a string to a double e.g:
```c
char *valueString = "12.23232";
double value = strtod(valueString, NULL);
```
}

The problem is an `undefined reference to 'strtod'`.
Included libraries:
```c
#include <assert.h>
#include <stdint.h>
#include <stdio.h>
#include <stdlib.h>
#include <string.h>
```

Neutree, 2018-06-22:
atof instead

cryarchy, 2018-06-22:
also not defined.

Neutree, 2018-06-22:
update to latest version please

cryarchy, 2018-06-22:
Thanks. This seems to be working now.

closed 2018-06-22 by cryarchy