Ai-Thinker-Open/GPRS_C_SDK #147: sprintf ignores `%f` format specifier

cryarchy, 2018-06-22:
## 1. SDK version（SDK 版本）
{
V2.107
}

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）
{
To reproduce the problem, use `sprintf ` with the `%f` format specifier. e.g:
```c
sprintf(buffer, "%f", 11.334455);
UART_Write(UART1, buffer, strlen(buffer)); // outputs "%f"
```
I have tested it with *strings* and *integers* and they all seem to be working as expected except *doubles*.
}

punee995, 2018-06-23:
pleas see issue : #72

punee995, 2018-06-23:
another way is ...
`sprintf(buffer, "%s", gcvt(11.334455));`

cryarchy, 2018-06-23:
@punee995 Thanks again.

For future reference, below is an example of the correct usage of `gcvt`:

```c
...
double latitude = 2236.3934, longitude = 11350.3831;
char coordBuffer[11];
...
fprintf("Latitude: %s\n", gcvt(latitude, 10, coordBuffer));         //-> Latitude: 2236.3934
fprintf("Longitude: %s\n", gcvt(longitude, 10, coordBuffer));   //-> Longitude: 11350.3831
```

Neutree, 2018-06-29:
support %f in version 2.108 now

closed 2018-06-29 by Neutree