Ai-Thinker-Open/GPRS_C_SDK #149: Cannot merge elf

cryarchy, 2018-06-23:
## 1. SDK version（SDK 版本）
{
V2.107
}

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）
{
The problem occurs during the building phase of a project. I am unsure as to how to reproduce the problem. However, I have utilized [this JSON parsing and generating library](http://ccodearchive.net/info/json.html) with the following modifications:

1. I have replaced all occurrences of `strtod` with `atof`.
2. I have included the `<api_debug.h>` header in *json.c* and replaced all occurrences of `printf(...)` with `trace(...)`.

When I run `build ontago2`, I get the following output:
```
build folder exist
number of processors: 4
Build host is WINDOWS

MAKE                  init

MAKE                  libs

MAKE                  libs/gps

MAKE                  libs/gps/minmea

PREPARING         libgps_debug.a

MAKE                  libs/utils

PREPARING         liblibs_debug.a

MAKE                  ontago2

LDGEN by csdk    cust.ld
LD                       ontago2.elf
ElfCombine          Elf binary & map file

GEN                   stripped (rm syms) elf file ontago2.elf
/hex/ontago2/ontago2.elf: in function '__nedf2':
/home/lee/gnu_src/gcc_build/mips-elf/libgcc/../.././gcc/dp-bit.c:1214: multiple definition of '__nedf2'
mips-elf-ld.exe: Dwarf Error: Can't find .debug_ranges section.
mips-elf-ld.exe: Dwarf Error: Can't find .debug_ranges section.
mips-elf-ld.exe: Dwarf Error: Can't find .debug_ranges section.
mips-elf-ld.exe: Dwarf Error: Can't find .debug_ranges section.
mips-elf-ld.exe: Dwarf Error: Can't find .debug_ranges section.
mips-elf-ld.exe: Dwarf Error: Can't find .debug_ranges section.
mips-elf-ld.exe: Dwarf Error: Can't find .debug_ranges section.
mips-elf-ld.exe: Dwarf Error: Can't find .debug_ranges section.
mips-elf-ld.exe: Dwarf Error: Can't find .debug_ranges section.
mips-elf-ld.exe: Dwarf Error: Can't find .debug_ranges section.
mips-elf-ld.exe: Dwarf Error: Can't find .debug_ranges section.
mips-elf-ld.exe: Dwarf Error: Can't find .debug_ranges section.
mips-elf-ld.exe: Dwarf Error: Can't find .debug_ranges section.
mips-elf-ld.exe: Dwarf Error: Can't find .debug_ranges section.
mips-elf-ld.exe: Dwarf Error: Can't find .debug_ranges section.
mips-elf-ld.exe: Dwarf Error: Can't find .debug_ranges section.
mips-elf-ld.exe: Dwarf Error: Can't find .debug_ranges section.
mips-elf-ld.exe: Dwarf Error: Can't find .debug_ranges section.
mips-elf-ld.exe: Dwarf Error: Can't find .debug_ranges section.
mips-elf-ld.exe: Dwarf Error: Can't find .debug_ranges section.
mips-elf-ld.exe: Dwarf Error: Can't find .debug_ranges section.
mips-elf-ld.exe: Dwarf Error: Can't find .debug_ranges section.
mips-elf-ld.exe: Dwarf Error: Can't find .debug_ranges section.
mips-elf-ld.exe: Dwarf Error: Can't find .debug_ranges section.
mips-elf-ld.exe: Dwarf Error: Can't find .debug_ranges section.
mips-elf-ld.exe: Dwarf Error: Can't find .debug_ranges section.
mips-elf-ld.exe: Dwarf Error: Can't find .debug_ranges section.
mips-elf-ld.exe: Dwarf Error: Can't find .debug_ranges section.
mips-elf-ld.exe: Dwarf Error: Can't find .debug_ranges section.
[... 250 lines cut ...]
mips-elf-ld.exe: Dwarf Error: Can't find .debug_ranges section.
mips-elf-ld.exe: Dwarf Error: Can't find .debug_ranges section.
hex/ontago2/SW_V2107_csdk.elf:/home/lee/gnu_src/gcc_build/mips-elf/libgcc/../../
./gcc/dp-bit.c:1214: first defined here
Cannot merge elf

CP                AUDIO CFP FILE......
AUDIO CFG         No audio cfp file
SREC              ontago2.srec for flash/romulator

LODTOBIN  C:/GPRS_C_SDK_V2.107/hex/ontago2/ontago2_flash.lod
LODTOBIN          Sucessful
LODCOMBINE        Combine user lod with Platform lod
LODCOMBINE        Combine sucessful

ROM    total:1048576 Bytes     used:10016 Bytes
RAM    total:1048576 Bytes     used:1088 Bytes
=== Build Time: 19s at 23/06/2018  5:45:31.70 ===
_____________________________________________________

C:\GPRS_C_SDK_V2.107>
```

**Footnote:** I am aware of the existence of the *jsmn* JSON parsing library (works fine). I would, however, prefer to use a JSON generator library as my code does a lot of dynamic JSON generation.
}

cryarchy, 2018-06-23:
I have narrowed down the cause of the fail to the existence of a `Trace` statement within an `if` block e.g:
```c
// is called on the API_EVENT_ID_SOCKET_RECEIVED event
void handle_response(char *response, double lat, double lng) {
    ...
    double latitude = 0, longitude = 0;
    ...
    for (i = 0; i < noOfTokens; i++) {
        // parse JSON response
    }
    /*---------- The Clause causing Problems ----------*/
    if (latitude == lat && longitude == lng) {
        Trace(1, "Device Location Update Successful\n");
        Trace(1, "Latitude: %f\n", longitude);
        Trace(1, "longitude: %f\n", longitude);
    } else {
        Trace(1, "Server and Local Device Location data Mismatch\n");
        Trace(1, "Latitude: %f != %f\n", lat, longitude);
        Trace(1, "longitude: %f != %f\n", lng, longitude);
    }
}
```
If I modify the above code to:
```c
// is called on the API_EVENT_ID_SOCKET_RECEIVED event
void handle_response(char *response, double lat, double lng) {
    ...
    double latitude = 0, longitude = 0;
    ...
    for (i = 0; i < noOfTokens; i++) {
        // parse JSON response
    }
    /*---------- The Clause causing Problems ----------*/

    Trace(1, "Device Location Update Successful\n");
    Trace(1, "Latitude: %f\n", longitude);
    Trace(1, "longitude: %f\n", longitude);
}
```
The project builds without any error. This has led me to the conclusion is not caused by the aforementioned JSON library as I was able to reproduce the same issue using *jsmn* as my JSON parsing library.

**NOTE:** The same problem occurred when I tried to replace the `Trace` statements with `UART_Write(UART1, ...)` statements.

Neutree, 2018-06-24:
there's multiple definition of `__nedf2` according to the error, add this word to file [platform_symbols_to_strip](https://github.com/Ai-Thinker-Open/GPRS_C_SDK/blob/master/platform/compilation/platform_symbols_to_strip)

cryarchy, 2018-06-25:
@Neutree Thanks! It worked.

closed 2018-06-25 by cryarchy