Ai-Thinker-Open/GPRS_C_SDK #152: How to get lat,lon in "degrees" V2.107

punee995, 2018-06-27:
## 1. SDK version（SDK 版本）

{

V2.107

}

---

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）

{
How to Get degree output in v2.107

```
latitude = gpsInfo->rmc.latitude.value/gpsInfo->rmc.latitude.scale;
longitude = gpsInfo->rmc.longitude.value/gpsInfo->rmc.longitude.scale;

gcvt(latitude,4,buff1);
gcvt(longitude,4,buff2);

Trace(1,"MINMEA : (%s,%s)",buff1,buff2);
```

(sorry does not have valid gps at this time)
and output is (FOR lat:9000.0000 , lon:00000.0000) :

    `MINMEA : (9000.0000,0.0000)`

and required output is 

    `MINMEA : (9000.0000,00000.0000)`

}

---

Neutree, 2018-06-27:
https://github.com/Ai-Thinker-Open/GPRS_C_SDK/blob/master/demo/gps/src/demo_gps.c#L137

punee995, 2018-06-27:
yes i'm aware of that but i need **degrees** output like **ddmm.mmmm** so how can i do that in **full scale** output.

punee995, 2018-06-27:
Solved!

closed 2018-06-27 by punee995

Neutree, 2018-06-27:
gcvt may can not do that,
you can write your own function,
or use sprintf to achive that

punee995, 2018-06-27:
i think minmea_rescale() will do!