Ai-Thinker-Open/GPRS_C_SDK #158: USSD API Problem

gsmwhatnot, 2018-07-06:
## 1. SDK version（SDK 版本）

{

2.108

}

---

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）

{

In `demo_ussd.c` I only changed `USSD_STRING` from `*#21#` into `*250*11#` which is to check SIM balance. But while sending USSD , problem occurs at the event `API_EVENT_ID_USSD_SEND_SUCCESS` while running following line `int len = GSM_8BitTo7Bit(result->usdString,buffer,result->usdStringSize);`

Problem is that chip goes into debug loop 
`Detected XCPU in the GDB loop (0x9db00000), connect GDB debugger for more info.`
`Detected BCPU in the GDB loop (0x9db10000), connect GDB debugger for more info.`
Also I can't connect to chip via GDB debugger, it shows error `Failed to detect chip`

I commented out the line and program went through so I expected to see event `API_EVENT_ID_USSD_IND` happens ( In response to my `*250*11#` code ) but that event has never been called !

This seems some kind of bug to me.
}

---

gsmwhatnot, 2018-07-08:
Ok, it seems that my operator is sending USSD response in UCS-2 format. So passing that character code into `GSM_8BitTo7Bit` will cause the gdb loop.

Please review `GSM_8BitTo7Bit` and `GSM_7BitTo8Bit` functions for sanitation.

Neutree, 2018-07-09:
https://github.com/Ai-Thinker-Open/GPRS_C_SDK/blob/master/include/api_charset.h#L9

here's the UCS2 to local

closed 2018-07-09 by gsmwhatnot