Ai-Thinker-Open/GPRS_C_SDK #161: Post http data as curl

updesh31, 2018-07-11:
## 1. SDK version（SDK 版本）

{

SW_V2110_csdk

}

---

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）

{

how can I post data as curl?

    curl -X POST \
      'http://xx.xx.xx.xx:7896/iot/d?k=4jggokgpepnvsb2uv4s40d59ov&i=deviceSOS3' \
      -H 'content-type: text/plain' \
      -d 'sos|Problem'

where -d is for data.

}

---

Neutree, 2018-07-12:
https://github.com/Ai-Thinker-Open/GPRS_C_SDK/blob/master/demo/gps_agps/src/demo_agps.c#L193

Neutree, 2018-07-12:
http is application layer protocol, and based on TCP/IP, so we can implement http by socket(TCP)

updesh31, 2018-07-13:
Thanks it's working.

closed 2018-07-13 by updesh31