Ai-Thinker-Open/GPRS_C_SDK #162: Combining two codes in CSDK

Tanukaroy4494, 2018-07-13:
## 1. SDK version（SDK 版本）

{

V2.108

}

---

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）

{

Hi, I am trying to combine two codes i.e. mqtt_ssl and gps. I would like to know if i can pass two parameters to  OS_SetUserMainHandle() like this  **OS_SetUserMainHandle(&mainTaskHandle,&gpsTaskHandle)**
If not then how can i make two event dispatch to work together?

}

---

updesh31, 2018-07-13:
you should create two threads.
ex.
OS_SetUserMainHandle(&mainTaskHandle);
OS_SetUserMainHandle(&gpsTaskHandle);

Tanukaroy4494, 2018-07-19:
@updesh31 
This process does not create two threads instead replaces the first one with the second.

closed 2018-07-24 by Tanukaroy4494