Ai-Thinker-Open/GPRS_C_SDK #163: watchdog function

updesh31, 2018-07-13:
## 1. SDK version（SDK 版本）

{

SW_V2110_csdk

}

---

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）

{

Can I use watchdog? If yes how?Because it's a very important function for real-time applications so I want to use it.

}

---

gsmwhatnot, 2018-07-16:
I believe RDA8955 does not have any `watchdog` capability ( At least I am not aware of one ). It only has this `sxs_Raise` function for their internal development/debugging. So any time something goes wrong inside the chip, this function get called and you can see debugger loop message in cooltools.

But here is this workaround I was thinking about that while back but didn't actually implement that. You can hook up HTx and HRx to a free UART port and read and write messages from module just like how cooltools do that. You can pretty much detect GDB loop, restart module, read events that are not still implemented in SDK version. A little bit of sniffing data may give you an idea how the protocol works.

punee995, 2018-07-17:
Use external watchdog timer IC.!

Neutree, 2018-07-17:
watch dog activated in release but debug version not

updesh31, 2018-07-18:
Ok I will test it.