Ai-Thinker-Open/GPRS_C_SDK #164: Critical TCP communication problem

updesh31, 2018-07-13:
## 1. SDK version（SDK 版本）

{

SW_V2110_csdk

}

---

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）

{
My gps tracker is working well but after 2-3 hours it's stop working and after rebooting it's again start working.I am getting this problem from starting. Today I debug it with trace tool and get attached log:
It shows that after detach from network module is not able to activate gprs communication.

[gpstracker.txt](https://github.com/Ai-Thinker-Open/GPRS_C_SDK/files/2191984/gpstracker.txt)

}

---

updesh31, 2018-07-13:
I think I get the problem. The problem starts when module detached from network and again goes to activate context.

 [00:52:41.058]   MMI 02 : network register success
 [00:52:41.058]   MMI 01 : attach status:0
 [00:52:41.058]   MMI 02 : active status:0
 [00:52:43.070]   MMI 02 : active status:0
 [00:52:43.070]   MMI 02 : network attach success
 [00:52:43.491]   MMI 02 : network got time
 [00:52:44.412]   MMI 02 : network activate success
Now problem starts:
[00:53:17.129]   MMI 02 : network deactived
[00:53:17.612]   MMI 02 : network detached
[00:53:19.562]   MMI 02 : active status:0
[00:53:19.562]   MMI 02 : network attach success
[00:53:19.968]   MMI 02 : network got time
[00:53:20.966]   MMI 02 : network activate failed
[00:53:21.076]   MMI 02 : network deactived
[00:54:07.551]   MMI 02 : network detached
[00:54:08.947]   MMI 02 : active status:0
[00:54:08.947]   MMI 02 : network attach success
[00:54:09.150]   MMI 02 : network got time
[00:54:09.212]   MMI 02 : network activate failed
[00:54:09.228]   MMI 02 : network deactived
[00:54:12.553]   MMI 02 : signal quality:14
[00:54:17.655]   MMI 02 : signal quality:15
[00:54:27.654]   MMI 02 : signal quality:14
[00:54:42.656]   MMI 02 : signal quality:15
[00:54:57.632]   MMI 02 : network detached
[00:54:59.223]   MMI 02 : active status:0
[00:54:59.223]   MMI 02 : network attach success
[00:54:59.722]   MMI 02 : network got time
[00:54:59.753]   MMI 02 : network activate failed

updesh31, 2018-07-16:
Waiting for your reply?

gsmwhatnot, 2018-07-16:
Sharing your code may help get to the point/problem much faster. I am not having problem with GPRS in my code.

updesh31, 2018-07-17:
I will share my code.But what should I do if I received network API_EVENT_ID_NETWORK_DEACTIVED?

updesh31, 2018-07-17:
I have attached my code.
[ltcs.zip](https://github.com/Ai-Thinker-Open/GPRS_C_SDK/files/2200665/ltcs.zip)

updesh31, 2018-07-18:
Any suggestion?

punee995, 2018-07-18:
Not tested
`case API_EVENT_ID_NETWORK_DEACTIVED:
	bool rst = false;
	if(!Network_StartDeactive(context))
		Trace(1,"Failed!");
	for(int i = 0; i < 5; i++){
		if(!Network_StartActive(context))
			Trace(1,"Failed!");
			rst = true;
	}
	if(rst)
		PM_Restart();`

updesh31, 2018-07-18:
It's the case where we need to reboot the module. Case repeats every 2-3 hrs. Is there any permanent solution or I should reboot the module every 2-3 hrs?

updesh31, 2018-07-18:
In at command mode 
If(+CGATT==0)
 AT+CGATT=1
AT+CIPSHUT
Then we again start the GPRS activation.
That's work. Then why error in SDK?

punee995, 2018-07-18:
what i'm trying to say is **Network_StartDeactive(context)** and  **Network_StartDetach()** and then 
**Network_StartActive(context)** and **Network_StartAttach()** again!

updesh31, 2018-07-18:
Thanks I am testing it. 🙂

updesh31, 2018-07-18:
One more help in function: 
Network_StartDeactive(context)

What is context? How will I get it?

updesh31, 2018-07-18:
In sdk_init.h it is given that
    bool                (*Network_StartActive)(Network_PDP_Context_t context);
    bool                (*Network_StartDeactive)(uint8_t contextID);

punee995, 2018-07-18:
context is struct of apn, username and password.

also i may think you have problem with RAM overflow A9G crashes when RAM overflow occures. so try clearing variables using OS_Free();

updesh31, 2018-07-18:
Context is struct ok but in header file it is given that only uint8_t contextID is only argument for Network_StartDeactive();
A9G crash is not the problem.

updesh31, 2018-07-19:
Still waiting for your reply...... ☺

closed 2018-07-28 by Neutree

updesh31, 2018-08-06:
not getting success to start again gprs service.

Johhnzero, 2019-04-15:
I have the same problem，have you solved

medida, 2019-11-25:
to get the device to reconnect to the network in FULL - you need to overwrite the APN details and then call again.
   Network_PDP_Context_t context = {
                .apn        ="somethingwrongtomakeitfail",
                .userName   = "",
                .userPasswd = ""
            };
            Network_StartActive(context);
          
            Network_PDP_Context_t context2 = {
                .apn        ="mobiledata",
                .userName   = "",
                .userPasswd = ""
            };
         
            Network_StartActive(context2);
           //   Network_SetFlightMode(false); // worked when here
            OS_Sleep(3000);