Ai-Thinker-Open/GPRS_C_SDK #165: HOW TO COMBINE TWO CODES to send gps data to BROKER

fewtechiot, 2018-07-16:
**HELLO THERE,**

I'm Trying to send GPS data through MQTT to the broker `test.mosquitto.org`,
can anyone tell me how to combine these codes, how can I send variable GPS data in PUBLISH_PAYLOAD
CAN ANYONE GUIDE ME?
@Neutree 
**THANK YOU.**

NathasLab, 2018-07-19:
I have tried the same, but had no succes. 
here is my code:

```
#include "stdbool.h"
#include "stdint.h"
#include "stdio.h"
#include "string.h"

#include "api_os.h"
#include "api_debug.h"
#include "api_event.h"
#include "api_mqtt.h"
#include "api_network.h"
#include "api_socket.h"

#include <string.h>
#include <api_gps.h>
#include <api_event.h>
#include <api_hal_uart.h>
#include "gps_parse.h"
#include "math.h"
#include "gps.h"

#define BROKER_IP  "scheufele.ddns.net"
#define BROKER_PORT 1883
#define CLIENT_ID  "gprs"
#define CLIENT_USER ""
#define CLIENT_PASS ""
#define SUBSCRIBE_TOPIC "app"
#define PUBLISH_TOPIC   "gps"
#define PUBLISH_INTERVAL 5000

#define MQTT_TASK_STACK_SIZE    (2048 * 2)
#define MQTT_TASK_PRIORITY      2
#define MQTT_TASK_NAME          "Main Test Task"

#define SECOND_TASK_STACK_SIZE    (2048 * 2)
#define SECOND_TASK_PRIORITY      3
#define SECOND_TASK_NAME          "MQTT Test Task"

#define GPS_TASK_STACK_SIZE    (2048 * 2)
#define GPS_TASK_PRIORITY      1
#define GPS_TAST_NAME          "GPS Test Task"

static HANDLE gpsTaskHandle = NULL;
bool flag = false;
bool isGpsOne = true;

static HANDLE mainTaskHandle = NULL;
static HANDLE secondTaskHandle = NULL;

static HANDLE semMqttStart = NULL;

char publish_payload[300] = "Hier wird die GPS-Information stehen.";
char buffer[300];

typedef enum{
    MQTT_EVENT_CONNECTED = 0,
    MQTT_EVENT_DISCONNECTED ,
    MQTT_EVENT_MAX
}MQTT_Event_ID_t;

typedef struct {
    MQTT_Event_ID_t id;
    MQTT_Client_t* client;
}MQTT_Event_t;

typedef enum{
    MQTT_STATUS_DISCONNECTED = 0,
[... 398 lines cut ...]
src/demo_mqtt.c:232: warning: format '%f' expects type 'double', but argument 12 has type 'struct minmea_float'
src/demo_mqtt.c: In function 'OnTimerPublish':
src/demo_mqtt.c:342: error: incompatible types when assigning to type 'char[300]' from type 'char *'
src/demo_mqtt.c:343: error: request for member 'member' in something not a structure or union
src/demo_mqtt.c:343: error: request for member 'member' in something not a structure or union
src/demo_mqtt.c: In function 'mqtt_Main':
src/demo_mqtt.c:431: warning: implicit declaration of function 'delay'
C:/GPRS_C_SDK/platform/compilation/cust_rules.mk:958: recipe for target 'C:/GPRS_C_SDK/build/mqtt/demo/mqtt/obj/debug/de
mo_mqtt.o' failed
make[3]: *** [C:/GPRS_C_SDK/build/mqtt/demo/mqtt/obj/debug/demo_mqtt.o] Error 1
make[3]: *** Waiting for unfinished jobs....
C:/GPRS_C_SDK/platform/compilation/cust_rules.mk:975: recipe for target 'dependencies' failed
make[2]: *** [dependencies] Error 2
C:/GPRS_C_SDK/platform/compilation/cust_rules.mk:718: recipe for target 'C:/GPRS_C_SDK/build/mqtt//mqtt.srec' failed
make[1]: *** [C:/GPRS_C_SDK/build/mqtt//mqtt.srec] Error 2
C:/GPRS_C_SDK/platform/compilation/cust_rules.mk:371: recipe for target 'lod' failed
make: *** [lod] Error 2
ROM    total:1048576 Bytes     used:2656 Bytes
RAM    total:1048576 Bytes     used:32 Bytes
=== Build Time: 1s at 19.07.2018 17:30:22,72 ===
_____________________________________________________
PS C:\GPRS_C_SDK>

```
somehow it seems that there is a problem converting char* to char[].

fewtechiot, 2018-07-20:
Hello there,
Thanks...
But anyhow I want to make it work.
We can put our combined efforts to solve this!
If you don't mind. how can I reach you?
Do you have Skype ID or something?

Thanks for your response.

closed 2018-07-28 by Neutree

fewtechiot, 2018-07-28:
Hello,@neutree I already did this I'm able to send data to mosquitito server as well as to secure microsoft AZURE  platform...
Please don't close any issues if you can't help..!

😏🙄

Neutree, 2018-07-28:
please close you ticket if you  resolved your problem as you describe, and you not you can reopen it

ricardopera, 2018-12-05:
This code is working, It sends gps data to demo.thingsboard.com.

`#include "string.h"
#include "stdio.h"
#include "api_os.h"
#include "api_gps.h"
#include "api_event.h"
#include "api_hal_uart.h"
#include "api_debug.h"
#include "buffer.h"
#include "gps_parse.h"
#include "math.h"
#include "gps.h"
#include "stdbool.h"
#include "stdint.h"
#include "api_mqtt.h"
#include "api_network.h"
#include "api_socket.h"

#define MAIN_TASK_STACK_SIZE    (1024 * 2)
#define MAIN_TASK_PRIORITY      0
#define MAIN_TASK_NAME          "Main Task"

#define SECOND_TASK_STACK_SIZE    (1024 * 2)
#define SECOND_TASK_PRIORITY      1
#define SECOND_TASK_NAME          "Second Task"

#define BROKER_IP  "demo.thingsboard.io"
#define BROKER_PORT 1883
#define CLIENT_ID  "A9G GPRS"
#define CLIENT_USER "rieIPosjsTkdn6GbLFYs"
#define CLIENT_PASS "mqtt"
#define SUBSCRIBE_TOPIC "v1/devices/me/rpc/request/+"
#define PUBLISH_TOPIC   "v1/devices/me/telemetry"
#define PUBLISH_INTERVAL 5000 //10s
#define PUBLISH_PAYLOEAD "{\"temperature\":\"50\",\"humidity\":\"60\"}"

HANDLE mainTaskHandle  = NULL;
HANDLE secondTaskHandle = NULL;

//Variáveis MQTT
static HANDLE semMqttStart = NULL;
typedef enum{
    MQTT_EVENT_CONNECTED = 0,
    MQTT_EVENT_DISCONNECTED ,
    MQTT_EVENT_MAX
}MQTT_Event_ID_t;

typedef struct {
    MQTT_Event_ID_t id;
    MQTT_Client_t* client;
}MQTT_Event_t;

typedef enum{
    MQTT_STATUS_DISCONNECTED = 0,
    MQTT_STATUS_CONNECTED       ,
    MQTT_STATUS_MAX
}MQTT_Status_t;

MQTT_Status_t mqttStatus = MQTT_STATUS_DISCONNECTED;

// Variáveis GPS
bool flag = false;
bool isGpsOn = true;

void OnMqttReceived(void* arg, const char* topic, uint32_t payloadLen)
{
    Trace(1,"MQTT received publish data request, topic:%s, payload length:%d",topic,payloadLen);
}

[... 325 lines cut ...]

void AppMainTask(VOID *pData)
{
    API_Event_t* event=NULL;
    
    secondTaskHandle = OS_CreateTask(LoopTask ,
        NULL, NULL, SECOND_TASK_STACK_SIZE, SECOND_TASK_PRIORITY, 0, 0, SECOND_TASK_NAME);
        
    while(1)
    {
        if(OS_WaitEvent(mainTaskHandle, (void**)&event, OS_TIME_OUT_WAIT_FOREVER))
        {
            EventDispatch(event);
            OS_Free(event->pParam1);
            OS_Free(event->pParam2);
            OS_Free(event);
        }
    }
}
void app_Main(void)
{
    mainTaskHandle = OS_CreateTask(AppMainTask ,
        NULL, NULL, MAIN_TASK_STACK_SIZE, MAIN_TASK_PRIORITY, 0, 0, MAIN_TASK_NAME);
    OS_SetUserMainHandle(&mainTaskHandle);
}`

CallaDai, 2019-07-18:
does it works? I just meet the same problem, but I have no idea about it. I want to see some guides or tuturil about it. Can you help me ? or send you the message by Jabber? whats app?

CallaDai, 2019-07-18:
I have a question. when I run this code, it reports me the error: I don't understand these errors.
D:/GPRS_C_SDK_V2112/GPRS_C_SDK/build/testDai/init/lib\libinit_debug.a(sdk_init.o): In function `user_Main':
D:\GPRS_C_SDK_V2112\GPRS_C_SDK\init/src/sdk_init.c:30: undefined reference to `testDai_Main'
value:0, p:ffffffff88240018, addend:0, symbol:0
D:\GPRS_C_SDK_V2112\GPRS_C_SDK\init/src/sdk_init.c:30: relocation truncated to fit: R_MIPS16_26 against `testDai_Main'
D:/GPRS_C_SDK_V2112/GPRS_C_SDK/platform/compilation/cust_rules.mk:722: recipe for target 'D:/GPRS_C_SDK_V2112/GPRS_C_SDK/build/testDai//testDai.elf' failed
make[2]: *** [D:/nuvelos/GPRS_C_SDK_V2112/GPRS_C_SDK/build/testDai//testDai.elf] Error 1
D:/GPRS_C_SDK_V2112/GPRS_C_SDK/platform/compilation/cust_rules.mk:718: recipe for target 'D:/GPRS_C_SDK_V2112/GPRS_C_SDK/build/testDai//testDai.srec' failed
make[1]: *** [D:/GPRS_C_SDK_V2112/GPRS_C_SDK/build/testDai//testDai.srec] Error 2
D:/GPRS_C_SDK_V2112/GPRS_C_SDK/platform/compilation/cust_rules.mk:371: recipe for target 'lod' failed
make: *** [lod] Error 2

CRONOS-IOS, 2020-04-24:
> void StartTimerPublish (intervalo uint32_t, cliente MQTT_Client_t *)
> {
> OS_StartCallbackTimer (mainTaskHandle, intervalo, OnTimerPublish, (void *) cliente);
> }
> 
> void SecondTaskEventDispatch (MQTT_Event_t * pEvent)
> {

no compila el codigo....

mirekhk, 2021-05-05:
This one will be compiled:-)
`#include "string.h"
#include "stdio.h"
#include "api_os.h"
#include "api_gps.h"
#include "api_event.h"
#include "api_hal_uart.h"
#include "api_debug.h"
#include "buffer.h"
#include "gps_parse.h"
#include "math.h"
#include "gps.h"
#include "stdbool.h"
#include "stdint.h"
#include "api_mqtt.h"
#include "api_network.h"
#include "api_socket.h"

#define MAIN_TASK_STACK_SIZE (1024 * 2)
#define MAIN_TASK_PRIORITY 0
#define MAIN_TASK_NAME "Main Task"

#define SECOND_TASK_STACK_SIZE (1024 * 2)
#define SECOND_TASK_PRIORITY 1
#define SECOND_TASK_NAME "Second Task"

#define BROKER_IP "demo.thingsboard.io"
#define BROKER_PORT 1883
#define CLIENT_ID "A9G GPRS"
#define CLIENT_USER "rieIPosjsTkdn6GbLFYs"
#define CLIENT_PASS "mqtt"
#define SUBSCRIBE_TOPIC "v1/devices/me/rpc/request/+"
#define PUBLISH_TOPIC "v1/devices/me/telemetry"
#define PUBLISH_INTERVAL 5000 //10s
#define PUBLISH_PAYLOEAD "{"temperature":"50","humidity":"60"}"

HANDLE mainTaskHandle = NULL;
HANDLE secondTaskHandle = NULL;

//Variáveis MQTT
static HANDLE semMqttStart = NULL;
typedef enum{
MQTT_EVENT_CONNECTED = 0,
MQTT_EVENT_DISCONNECTED ,
MQTT_EVENT_MAX
}MQTT_Event_ID_t;

typedef struct {
MQTT_Event_ID_t id;
MQTT_Client_t* client;
}MQTT_Event_t;

typedef enum{
MQTT_STATUS_DISCONNECTED = 0,
MQTT_STATUS_CONNECTED ,
MQTT_STATUS_MAX
}MQTT_Status_t;

MQTT_Status_t mqttStatus = MQTT_STATUS_DISCONNECTED;

// Variáveis GPS
bool flag = false;
bool isGpsOn = true;

void OnMqttReceived(void* arg, const char* topic, uint32_t payloadLen)
{
Trace(1,"MQTT received publish data request, topic:%s, payload length:%d",topic,payloadLen);
}

void OnMqttReceiedData(void* arg, const uint8_t* data, uint16_t len, MQTT_Flags_t flags)
[... 339 lines cut ...]
}
}
void app_Main(void)
{
mainTaskHandle = OS_CreateTask(AppMainTask ,
NULL, NULL, MAIN_TASK_STACK_SIZE, MAIN_TASK_PRIORITY, 0, 0, MAIN_TASK_NAME);
OS_SetUserMainHandle(&mainTaskHandle);
}
`

This is compile result:
`[LOD]               D:/Documents/AiThinker/GPRS_C_SDK/hex/app/app_flash.lod
[LODCOMBINE]        Combine with Platform lod
[LODCOMBINE]        Combine sucessful

-------------------------------------------------
ROM    total:1048576 Bytes     used:17984 Bytes
RAM    total:1048576 Bytes     used:5168 Bytes
=================================================
Start Time : 16:11:52.42
End   Time : 16:12:16.39
Build Time : 23.997s
=================================================`

Thx a lot Riccardo, It explains a lot.