Ai-Thinker-Open/GPRS_C_SDK #166: 'sed' is not recognized

seym45, 2018-07-16:
SDK version: 2.108

when I run ./build.bat app in windows 10 powershell, the output is 
 
GPRS_C_SDK> .\build.bat app
CSDTK PATH: F:\GPRS_C_SDK
'sed' is not recognized as an internal or external command,
operable program or batch file.
'sed' is not recognized as an internal or external command,
operable program or batch file.
build folder exist
number of processors: 4

Can anybody help me with installation, please?

gsmwhatnot, 2018-07-16:
According to this section https://ai-thinker-open.github.io/GPRS_C_SDK_DOC/en/c-sdk/installation.html#installation double check that you have run as administrator this file `config_env_admin.bat`. It is better to restart your windows to make changes effective.
Also check in your Windows path if it is pointed to the `CSDTK4.2` folder. I believe you can run this command inside power shell to get you environment path : `echo %PATH%`

seym45, 2018-07-17:
Thank you, after the restart, It worked, thank you 👍 💯

closed 2018-07-17 by seym45