Ai-Thinker-Open/GPRS_C_SDK #169: compile problem wiyh sqrt()

updesh31, 2018-07-22:
## 1. SDK version（SDK 版本）

{

SW_V2110_csdk

}

---

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）

{

sqrt(2); //compiled without any error

but 
double x = 2.56;

sqrt(x); //compiler error
undefined reference to `sqrt'

}

---

Neutree, 2018-07-28:
I tried code with release 2.1.12
```
        double x = 2.56;

        double r = sqrt(x); //compiler error
        Trace(1,"r:%f",r);
```
the result is:
[image]

that's perfectly correctly

closed 2018-07-28 by Neutree

updesh31, 2018-07-28:
double x= 2.56;
        sqrt(x);
working but:
        double x= y;  //y is a double variable
        sqrt(x);
is not compiled. Same error undefined reference to `sqrt'. :(

gsmwhatnot, 2018-07-28:
I tried this code : 

`double y = 1.56;`
`double x = y; //y is a double variable`
`Trace(1,"(DBG) TT: %f", sqrt(x));`

and output : 
`(DBG) TT: 1.249000`

Please double check that you are using latest release v2.112 and latest CSDK4.2
If not please refer to this link : 
https://github.com/Ai-Thinker-Open/GPRS_C_SDK/releases

updesh31, 2018-07-28:
Please try with this:
		LIS3DSH_ReadACC(Buffer);
		x_Buff = Buffer[0];
		y_Buff = Buffer[1];
		z_Buff = Buffer[2];      
		roll = atan2(y_Buff , z_Buff) * 57.3;
               xyz = y_Buff * y_Buff + z_Buff * z_Buff;
		pitch = atan2((- x_Buff) , sqrt(xyz)) * 57.3;

I am using v2.112 and latest CSDK4.2.

gsmwhatnot, 2018-07-28:
Since I don't know the types of `x_Buff`, `y_Buff` and `z_Buff` so `xyz` type is unknown as well. Please try the following line and let me know the results : 
`pitch = atan2((- x_Buff) , sqrt((double)xyz)) * 57.3;`
also this one : 
`pitch = atan2((- x_Buff) , sqrt((float)xyz)) * 57.3;`

updesh31, 2018-07-28:
oh sorry all variables are of type double.

I will check your suggestion.

updesh31, 2018-07-28:
pitch = atan2((- x_Buff) , sqrt((double)xyz)) * 57.3;
also this one :
pitch = atan2((- x_Buff) , sqrt((float)xyz)) * 57.3;

Not working :(

gsmwhatnot, 2018-07-28:
Ok updesh31, I was going mad because of your situation ... I finally found what is the problem. Why we didn't get error while compiling with `sqrt(4)` ? Because as part of optimizations compiler calculates the sqrt(4) and put it in the variable. Because this is a constant value. In modern compilers this is done automatically by the compiler but in your situation you are passing a variable to sqrt() so math.h library has to be linked to your file with -lm ( -LM ) option during compilation. I tried adding this flag inside `Makefile` but I didn't have any success.

I think this problem can be solved by Neutree as he/she may have better knowledge about the RDA compiler.

For further information read this link : 
https://stackoverflow.com/questions/3533594/sqrt-function-not-working-with-variable-arguments/3533614