Ai-Thinker-Open/GPRS_C_SDK #170: SIM Pin code

gamecompiler, 2018-07-22:
Where and how can i enter the SIM cards pin code in the RTOS code? 

(For example the demo mqtt code, my simcard got locked due to too many false attempts, i delocked it with my smartphone but the problem will still always happen).

Neutree, 2018-07-28:
there's no operation related to PIN code

gamecompiler, 2018-07-28:
So i have to remove the sim cards PIN authentification feature?

gamecompiler, 2018-07-29:
Should i close this issue or should i leave it open? It could be labeled as "enhancement" - i will subscribe to this issue to be informed.

gsmwhatnot, 2018-07-30:
Let's see what is Neutree's opinion.

Neutree, 2018-07-31:
We can add this feature but PIN code related operation is pretty dangerous for some user, and most of application don't need it, so we didn't add this feature.
So... you have to remove the sim cards PIN authentification feature currently

ivomilev, 2019-03-03:
Hi,

Do you plan to release RTOS API for SIM PIN code soon? Without the PIN, our app/device lacks of security. If we use external MCU and AT cmd interface, it can be easy sniffed the serial interface with cheap logic analyzers and obtain the code if the device is stolen.

benedykt, 2019-07-07:
Hi, Any progress with entering PIN code function? It seems to be pretty easy to develop. I would say this should be basic function as not all sim card can be / and should be without pin code  All developers are aware of danger. It is like fuel in cars, without it - it will not drive - but refueling is dangerous.

Please. Help.

rolonja, 2021-06-02:
i Think the PIN function its very very basic for security reasons.